arkypita/LaserGRBL #1011: How to skip axis accelerating / decelerating area on raster engraving?

lome64, 2020-05-22:
Dear all, 
doing some raster engraving on my new 3018pro CNC Laser, I noticed dark areas 
on beginning and end of each scan line. Obviously there is more laser energy brought into 
the material there. 
A closer inspection revealed that Lasergrbl switches on the laser exactly the same moment a scanline 
starts and also switches the laser off, exactly where a scanline ends.
That means, the laser is switched on, even when the axis is still accelerating and has not yet 
reached the final speed and also the laser is still on, when the axis decellerates to zero at the end of the scanline.
That results in darker edges.
See attached the file I used for testing.
[image]
The Result clearly shows the dark edges in X direction:
[image]

To overcome the problem I added a light grey background to the image, a little bigger than the image itself:
[image]
So the laser is switched on within the light grey border (which is light enough to leave no imprint)
and as the image begins, the axis already reached it's full speed resulting in an equal dark imprint.
[image]

So it seems there is still a little room for improvement in Lasergrbl.
Or did I just miss the correct setting?
I also have a CO2 laser machine with a completely different software. 
That machine adds accelerating and decellerating space by itself.
That would also be a nice feature in Lasergrbl!
Maybe just a numeric fiels to enter the measurement of a accelerating "border" to give the axis 
a chance to reach its final speed, before the laser is switched on. (or off at the end of the scanline)
Also of course that problem appears when parts of a scanline that are white are skipped in rapid speed.

So if there is a setting for this already I would appreciate a little hint.
If not, it would be fine ifthis feature would be added to Lasergrbl.

Thank you!
  Lothar

arkypita, 2020-05-22:
Use grbl v1.1 with Laser Mode enabled and M4

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

lome64, 2020-05-22:
arkypita, thank you for your suggestion!

The machine runs grbl 1.1, but I was not aware of M4 dynamic power scaling. 
That sounds promising. I'll have a look at it.

Thank's!
  Lothar

lome64, 2020-05-22:
Hmmm.... 
Tried it. 
And it seems it is willing to do the right thing, but with very little success:

[image]

In my oppinion it's even worse than constant laser power.
Same settings as in the opening message above. Just switched laser-on-command from M3 to M4
Are there more parameters to tune this?

Lothar

lome64, 2020-05-22:
P.S.: In the log of Lasergrbl I cannot find a $32=1 command. Does Lasergrbl send it implicitely?

lome64, 2020-05-22:
I apologize! 
The machine was a bit messed up after making he changes. 
It showed strange behaviour and finally did not respond at all.
After a power cycle and restart of the software it now works perfect with M4 and $32=1
This is the newest result:
[image]

I'm happy with that! 
Thank you!

arkypita, 2020-05-22:
$32 is a setting, not a gcode.
Once set, always set

http://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

lome64, 2020-05-22:
> $32 is a setting, not a gcode.

I see. Found that in the GRBL settings. 
Thank you for your fast and competent help!

Lothar

closed 2020-05-22 by lome64