arkypita/LaserGRBL #1012: Finer engraving

fdurville, 2020-05-22:
Would it be possible to increase the image resolution (Quality) to 50 lines/mm?
I am currently using a laser which has a spot size that can be adjusted down to about 20-micron diameter. Thus having 50 lines/mm would allow me to realize extremely fine engraving...

arkypita, 2020-05-22:
The current limit of 20 lines per millimeter is an arbitrary limit. It is easy for me to remove it.
Keep in mind that 20 lines / mm correspond to about 500 dpi which is a rather high resolution.

Increasing the resolution also means that LaserGRBL will need to work with larger files, therefore more memory and more CPU for the process, but as long as you work on maximum dimensions of 30 * 40 cm there should be no problems.

What laser do you have, out of curiosity?

fdurville, 2020-05-22:
The laser is a Q-switch fiber laser which I built myself.
Frederic

arkypita, 2020-05-26:
I did some tests on LaserGRBL enabling the possibility to use a higher resolution. The processing becomes very heavy because you have to imagine that 10cm x 10cm at 50 lines / mm become an image of 5000x5000 px, which at 32bit per pixel are about 100MB.

Considering that the image undergoes several steps this means a slower and heavier work in memory.

Nothing unmanageable for a good modern PC, but definitely oversized for the normal user.

Let me know if you want to try a customized version of LaserGRBL I can send you one that admits higher values.

fdurville, 2020-05-26:
Thank you very much for the response.
I certainly do understand the issue of larger files when dealing with large
images and higher res. And it also would increase proportionally the time
it takes to engrave the whole image.
In my case, the largest image I have engraved so far is about 3x3cm, and I
would like to explore only small images (1x1 cm max) with the higher res
(doing small engraving on small jewellery for example, my laser can engrave
on gold).
So yes, I would definitively love to try a customized version that can take
higher res values.
Frederic

fdurville, 2020-06-01:
> You can try this pre-release
> https://github.com/arkypita/LaserGRBL/releases/tag/v3.5.0
>
> I'd be curious to see some of your works

Thanks again for this update.
I just tried it, and it works quite well.
Here is the difference
<https://drive.google.com/open?id=1_NkvUWjl19VE4ofxgebzaSOk7sWZ0xN8>
between res 40lines/mm (on blue anodized) and 20lines/mm (on orange
anodized), as looked under microscope 20X.
The difference is not visible under naked eye, but is clearly visible under
magnification.
Frederic

arkypita, 2020-06-02:
Hi @fdurville 
I'm glad to know that the change had the desired effects.
What is the size of the rose in the photo?

I think that few users will be able to take advantage of this possibility because a very special and expensive laser and a very high resolution of the motors are required, however it is an additional function that I have enabled with little effort.

Out of curiosity, what is the resolution of your engines? What is the smallest step they can take?

Currently LaserGRBL produces gcode with two decimal numbers after the decimal point, therefore the positioning that LaserGRBL produces is in 10 micron steps. Do you think it could also be useful to increase to 3 decimal places?

Anyway thanks for the donation!

arkypita, 2020-06-02:
Errata: LaserGRBL already support 3 decimal places if needed

fdurville, 2020-06-02:
I understand that most users will not need that fine resolution... and I
really appreciate your making the change.
The resolution of my stage step-motor is 400/mm, this is a bit arbitrary as
it is easy to change the "micro-step" on the driver board, or in my case on
the CNC-shield board (I use an Arduino UNO with a CNC-shield) for each
motor.
Frederic

StuartB4, 2020-06-02:
Just wondered what stepper drivers you are using, and have you considered the Trinamic drivers, microstepping up to 256.
https://hackaday.com/2016/09/30/3d-printering-trinamic-tmc2130-stepper-motor-drivers-shifting-the-gears

fdurville, 2020-06-02:
I use "generic" (i.e., from china) A4988 drivers... (that is based on the
Allegro A4988 chip). And they can also be set to up to 256 micro-steps.
To my knowledge, all step motor drivers are very similar from an electrical
/ electronics standpoint, as there are only very few different chips, and
those chips differ only in their current handling (and heat dissipation)
capabilities.The ones you mention do have a neat heatsink to improve their
heat dissipation capabilities, which maybe useful in intensive usage (and
also improve their overall reliability).
FD

fdurville, 2020-06-02:
I just realized that what I made a mistake: the A4988 driver has only up to
16 micro-steps.
FD

StuartB4, 2020-06-03:
Ok. I use the DRV8825's, they can go up to 1/32 stepping. They can also run at a slightly higher voltage and current than the 4998's.