arkypita/LaserGRBL #1014: laser on

ronisegoly, 2020-05-23:
I have [laser module](https://www.sainsmart.com/products/sainsmart-blue-laser-module-kit-for-genmitsu-cnc-router-3018-mx3?_pos=1&_sid=7d14826d2&_ss=r)
Connected to [CNC machine ](https://www.sainsmart.com/collections/genmitsu-cnc/products/sainsmart-genmitsu-cnc-router-3018-prover-kit)
I can connecto the module from lasergrbl SW, jog the axis and load file
But I can't turn laser on/off by SW
If i use physical button it will be on but all the time of course so can't really engrave
Any idea?
Tried m3 m4 and laser flag 0 and 1

StuartB4, 2020-05-23:
Do you have this enabled?

[image]

ronisegoly, 2020-05-23:
Yes, I tried check and uncheck
BTW not sure it's related, when I connected the laser module I was left with one cable which was not used
See attached
[image]
[image]
[image]

StuartB4, 2020-05-23:
I think that lead is for if you have the 3 pin 12v, Gnd, PWM socket on your main controller board.

ronisegoly, 2020-05-24:
the socket on my controller is two pin
what's the functionality of the PWM connection asdie from setting laser on and changing it's level? If that's the only one then I suspect the connection is faulty

StuartB4, 2020-05-24:
The 3 pin PWM wire supplies the power and the PWM signal to a 3 pin laser, but your laser has a seperate power supply, so from what I can see on the images you have what you need with the seperate power and the 2 pin PWM lead. The black wire is just a ground on either the 2 pin or 3 pin lead.

StuartB4, 2020-05-24:
The only other thing I can think of is to ground both power supplies together.

ronisegoly, 2020-05-24:
thanks
waiting for supplier's support

ronisegoly, 2020-05-26:
issue resolved.
Initial spindle speed in config file was 10000

closed 2020-05-26 by arkypita

StuartB4, 2020-05-26:
Ok. :+1: