arkypita/LaserGRBL #1015: Help with Genmitsu le5040

MoTaVa, 2020-05-23:
Let me first say I am not a programmer or coder. I am a woodworker.
I bought a Genmitsu le5040 to enhance some of my projects and gifts.

On some of the pieces I burn I am getting lines between objects. See the picture. Can anyone help me understand how to avoid this?
[image]

StuartB4, 2020-05-23:
Usually it's because Laser mode is not enabled in GRBL configuration. 
Start LaserGRBL and click Connect. 
Click the Grbl menu and Grbl configuration and have a look down the list for $32. It should 
be set to 1.

Then set $30 to 255 if it's not already.
Click the Write button and Close.

MoTaVa, 2020-05-23:
Thanks for responding.
I entered the changes you suggested. The machine appears to be working a bit faster than it was. So far no errant lines. However, I can no longer focus the beam using the M3 command. Any insight?

StuartB4, 2020-05-24:
You can make a custom button that switches the laser on in low power mode to allow you to focus.

Start LaserGRBL and click connect.
Right click on an empty part of the bar at the bottom next to the little man icon and click Add custom button and then use this.

Make sure you are connected then use the button.
This button is only valid if you have GRBL 1.1.
$32=0 switches off laser mode. $32=1 switches laser mode back on.
M4 = Laser on. M5 = Laser off. S = Laser power setting.

Select Type: TwoStateButton.

Enter the code below in to GCode box 1:

$32=0
M4
S3

Enter the code below in to  GCode box 2:

$32=1
M5

Tool Tip: Laser ON (Laser Mode=OFF) / Laser OFF (Laser Mode=ON)

Enabled: Idle.

Click Save.

[image]

MoTaVa, 2020-05-29:
Thank you very much for your help.
The changes you suggested have made a huge difference in the quality and speed of my engravings.

Have you any other helpful advice for the novice? Tips or tricks? 
Any insight you feel like sharing would be most appreciated.
Thanks.

closed 2020-12-16 by arkypita