arkypita/LaserGRBL #1017: language translation

lianzaozi, 2020-05-24:
I want to make some contributions to the language translation of LASERGRBL software. What can I use to edit .resx files quickly? ? ?

Schwarzi07, 2020-05-24:
You can use notepad++ to edit the file

arkypita, 2020-05-26:
do you want to add a new translation or improve existing one?

Ozzybanan, 2020-06-07:
Hello. Can you translate the program into Polish or can you write the translation step by step. Greetings.

Ciao. Puoi tradurre il programma in polacco o puoi scrivere la traduzione passo dopo passo. Saluti.

gmmanonymus111, 2020-06-07:
@Ozzybanan You should [read this](https://github.com/arkypita/LaserGRBL/blob/master/TRANSLATING.md) to get started with translation.
Once you downloaded and installed the required programs, you can start translating. First, you need to open a `.resx` file, without any suffiexes (like hu_HU etc...)
In the example, we will open `SettingsForm.resx` with ResEx
[image]
[image]
Once opened, change it to your locale
[image]
If your locale is not present, yo can create it this way: 
[image]
Once your locale is available, you can start translating! Just simply fill the empty spaces on the right in your language. The left side shows the components name, while in the middle there is the english description/text to help you.
[image]
Once done, you can open another `.resx` file without any suffixes (or if you use the filter method, any file showing up). When all is done, you can create a Pull Request (PR) on github. I'll write a step-by-step guide for it, too.

Ozzybanan, 2020-06-08:
I have translated files into Polish. Thank you for the tips.
ConnectLogForm.pl-PL.resx
CustomButtonForm.pl-PL.resx
GrblConfig.pl-PL.resx
HotkeyManagerForm.pl-PL.resx
IssueDetectorForm.pl-PL.resx
MainForm.pl-PL.resx
PreviewForm.pl-PL.resx
ResumeJobForm.pl-PL.resx
RunFromPositionForm.pl-PL.resx
SaveOptionForm.pl-PL.resx
SettingsForm.pl-PL.resx

ResEx does not open files:
ExceptionManager.resx
JogForm.resx
LicenseForm.resx
SplashScreenForm.resx
Strings.resx

How do you check translated files in LaserGRBL, is everything OK?

LaserGRBL-3.5.0

gmmanonymus111, 2020-06-08:
You can see the remaining parts [here](https://github.com/arkypita/LaserGRBL/blob/master/TRANSLATING.md)

arkypita, 2020-06-08:
> I have translated files into Polish. Thank you for the tips.
> ConnectLogForm.pl-PL.resx
> CustomButtonForm.pl-PL.resx
> GrblConfig.pl-PL.resx
> HotkeyManagerForm.pl-PL.resx
> IssueDetectorForm.pl-PL.resx [...]

I's a good starting point, but there are other form inside subfolders like RasterConverter\ and SvgConverter\

> ResEx does not open files:
> ExceptionManager.resx
> JogForm.resx
> LicenseForm.resx
> SplashScreenForm.resx

These forms are set up as non-translatable

> Strings.resx

This should instead be translatable. What problem does it report?

> How do you check translated files in LaserGRBL, is everything OK?
> LaserGRBL-3.5.0

You have to wait for me to include your pull-request in the master version of LaserGRBL and to release an updated version. 

Since it is a new language I have to add some entry points in the code.

arkypita, 2020-06-08:
> EDIT: to check your translation, open LaserGRBL and change the language to Polish. Restart. Now you can check settings, GRBL settings etc to see if it works/looks good

No, it's not so simple.
In order to see the result it is necessary to compile with Visual Studio.

arkypita, 2020-06-08:
@Ozzybanan 

If possible, complete your work with resx files inside RasterConverter, SvgConverter, PSHelper, AutoUpdate folders.

I have fixed an issue inside Strings.resx file. Check out/merge my latest master commit (I am not able to show you the exact procedure) so you can edit it in ResEx.

At the end follow @gmmanonymus111 instruction to create a pull-request with your files.
I'll add them in the next version.

Ozzybanan, 2020-06-08:
ResEx already opens the file
Strings.resx

Ozzybanan, 2020-06-09:
I have sent a translation of the LaserGRBL files into Polish. Please enter the translation into the program.

arkypita, 2020-06-09:
@Ozzybanan download latest pre-release and let me know if it's all ok
https://github.com/arkypita/LaserGRBL/releases/tag/v3.5.2

Ozzybanan, 2020-06-11:
The translation works very well.
It's nice to work.

I would change two things:

1) The message comes out the window.

LaserGRBL\LaserGRBL\SettingsForm.pl-PL.resx

Is:

 <data name="label21.Text" xml:space="preserve">
    <value>Uwaga: ta funkcja działa tylko z wersją grbl; &gt; v1.1 z włączonym „trybem laserowym”. 
Jeśli użyjesz automatycznego chłodzenia ze starszą wersją grbl (lub z wyłączonym „trybem laserowym”), laser nie wyłączy się podczas przerw i wypali dziurę w twojej pracy!</value>

Should:

 <data name="label21.Text" xml:space="preserve">
    <value>Uwaga: ta funkcja działa tylko z wersją grbl; &gt; v1.1 z włączonym „trybem laserowym”. 
Jeśli użyjesz automatycznego chłodzenia ze starszą wersją grbl (lub z wyłączonym „trybem laserowym”), 
laser nie wyłączy się podczas przerw i wypali dziurę w twojej pracy!</value>

2) Change of word.

LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.resx

Is:

<data name="TpPreview.Text" xml:space="preserve">
    <value>Prapremiera</value>

Should:

<data name="TpPreview.Text" xml:space="preserve">
    <value>Podgląd zmian</value>

Thank you and best regards.

Ozzybanan, 2020-06-11:
Error

Ozzybanan, 2020-06-11:
[image]

Ozzybanan, 2020-06-11:
[image]

Ozzybanan, 2020-06-14:
Updated.
Thank you @arkypita and @ gmmanonymus111 for the step-by-step instruction.
Greetings.