arkypita/LaserGRBL #1019: Stray Lines when doing longer engraving sessions

broitblat, 2020-05-25:
I've had an issue with most (all?) of my longer engraving sessions (greater than 4 hours or so) where I seem to get a few random horizontal lines near the end of the session (see picture).  Does anyone have any thoughts on what might cause this and what I can do about it?
[image]

StuartB4, 2020-05-25:
Do you have $32 set to 1 in grbl configuration. If not, change it to 1 and change $30 to 255.

You may also be able to decrease the time it takes to engrave. These 2 pictures have exactly the same settings, but the Acceleration rate of the X and Y axis ($120 and $121) have different settings. 
That cuts the engraving time down significantly. There is a limit to how fast you can go though before you start loosing or skiping steps.

[image]

[image]

broitblat, 2020-05-26:
> Do you have $32 set to 1 in grbl configuration. If not, change it to 1 and change $30 to 255.
> 
> You may also be able to decrease the time it takes to engrave. These 2 pictures have exactly the same settings, but the Acceleration rate of the X and Y axis ($120 and $121) have different settings.
> That cuts the engraving time down significantly. There is a limit to how fast you can go though before you start losing or skipping steps.
> 

Thanks.  I did not, as it turns out, have Laser Mode set.  I will try it, again, with this setting as soon as I can commit the time for another longer session.

I have done some playing around with the acceleration rate settings, but will continue to experiment with that, too.

broitblat, 2020-05-26:
Oops, hadn't really meant to close this, just yet.

broitblat, 2020-05-30:
So, I've tried an image that had a stray line in it, twice, using the LaserMode setting.  Once, it completed without any stray lines, and once, it had a stray line (but in a different place than the original).

closed 2020-12-16 by arkypita