arkypita/LaserGRBL #1020: Size problems

Sughero77, 2020-05-26:
Dear all, I have a problem with size of images.
I select a dimension of image during the import section. When I start to engrave with laser, dimensions of image engraved is about 1,8-1,9 times smaller that what selected. This dimension is every time 1,8-1,9 times smaller.
I cannot find any button or selection to calibrate laser-software, can you help me?
Thanks
Marco

StuartB4, 2020-05-26:
Start LaserGRBL and click Connect.
Goto Grbl menu and click Grbl configuration.
Look down the list for $100 and $101.
Make a note of the numbers in the value boxes for both.
Then go to this website to calibrate the steps per mm.
https://jscalc.io/calc/tHjfIiKMKjDF42Hg?fbclid=IwAR219p6S0qWzVAsY8aLC8nyQaXaMpfa-mSTNCIXA7TAMPa3VzrGoCDcimsI

This is how you use it.

[image]

StuartB4, 2020-05-26:
When you have the new numbers make sure LaserGRBL is still connected and enter them in to the $100 and $101 value boxes in grbl configuration and click the Write button, then close.

arkypita, 2020-05-26:
http://lasergrbl.com/wp-content/uploads/2020/05/Configurazione-Grbl-ITA.pdf

Sughero77, 2020-05-27:
Thanks to all, freinds.
I will check in the evening and come back to you.
Thanks, great job

Sughero77, 2020-05-28:
Thanks to all, problem solved, I setted correctly the GRBL!!!

closed 2020-06-02 by arkypita