arkypita/LaserGRBL #1022: cant get my laser to work

WATSON-DOG, 2020-05-27:
hi I have just purchased a laser master2 engraver but I cant get the laser to work, I am very new to this, I have built it together, it moves but no red dot to show the beam? any ideas

thecuress, 2020-05-27:
which software use ?....and look if in grbl set you have set the laser to 1 in line $32 in grbl

WATSON-DOG, 2020-05-28:
I use laserGRBL and I still cant get it to work?

thecuress, 2020-05-28:
ok ..but in the $$ set you have $32 active?

WATSON-DOG, 2020-05-28:
it says in $$32 laser-mode enable , vaule 1, unit boolean, despription, enables laser mode consective G1/2/3 commants will not halt when spindle speed is changed

does this help?

thecuress, 2020-05-28:
which hardware use for command the motor ..laser ecc ecc

WATSON-DOG, 2020-05-28:
I am new to this so have no idea what it all means, I am an artist so wanted to do some engraving I have a laser master2 which I have just got but cant seem to get the laser to work

thecuress, 2020-05-28:
you have this ?  https://it.banggood.com/New-NEJE-Master-2-Desktop-CNC-Wood-Router-20W-Laser-Engraver-Cutter-Laser-Engraving-Machine-APP-Control-for-Windows-or-Mac-or-Android-p-1678509.html?gpla=1&gmcCountry=IT&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=xibei&utm_campaign=xibei-ssc-itg-label3-30-0408-19cov&gclid=EAIaIQobChMI8dbz4_TW6QIVSYfVCh2ffwpUEAQYAiABEgIA-_D_BwE&cur_warehouse=CN

WATSON-DOG, 2020-05-28:
images.githubusercontent.com/66023993/83167751-68e6fc00-a108-11ea-8f1d-50af4dd48fb2.jpg)

WATSON-DOG, 2020-05-28:
laser master 2 INGRAVER IS A DESKTOP LASER  THERE IS A G-SENSOR ON THE MOTHERBOARD

thecuress, 2020-05-28:
LINE $30 what value does it have?

thecuress, 2020-05-28:
or create a specif button http://lasergrbl.com/usage/custom-buttons/

WATSON-DOG, 2020-05-28:
line$30 as 1000

thecuress, 2020-05-28:
ok-.......try to make this,so it can create a button, and see if the laser goes or not

> read the whole page http://lasergrbl.com/usage/custom-buttons/

thecuress, 2020-05-28:
[image]

WATSON-DOG, 2020-05-28:
thank you I have added all the extra buttons but still not having the  beam working. it arm moves as if it is doing the lasering but no beam works. very frustrated any other ideas please

WATSON-DOG, 2020-05-29:
When using the button to turn on laser for focusing or laser blink On/Off it does work but after we load an image in it will reposition itself to that location begin the movements as if its lasering but the laser itself wont turn on at that point. As it turns on during the focus/blink I assume it's one of the settings in GBRL configuration this is what I have set.

$0 | Step pulse time | 20
-- | -- | --
$1 | Step idle delay | 50
$2 | Step pulse invert | 0
$3 | Step direction invert | 0
$4 | Invert step enable pin | 0
$5 | Invert limit pins | 0
$6 | Invert probe pin | 0
$10 | Status report options | 1
$11 | Junction deviation | 0.050
$12 | Arc tolerance | 0.002
$13 | Report in inches | 0
$20 | Soft limits enable | 1
$21 | Hard limits enable | 1
$22 | Homing cycle enable | 1
$23 | Homing direction invert | 7
$24 | Homing locate feed rate | 600.000
$25 | Homing search seek rate | 3000.000
$26 | Homing switch debounce delay | 100
$27 | Homing switch pull-off distance | 2.000
$30 | Maximum spindle speed | 1000
$31 | Minimum spindle speed | 1
$32 | Laser-mode enable | 1
$33 |   | 250.000
$100 | X-axis travel resolution | 80.000
$101 | Y-axis travel resolution | 80.000
$102 | Z-axis travel resolution | 80.000
$110 | X-axis maximum rate | 9000.000
$111 | Y-axis maximum rate | 9000.000
$112 | Z-axis maximum rate | 9000.000
$120 | X-axis acceleration | 2200.000
$121 | Y-axis acceleration | 1800.000
$122 | Z-axis acceleration | 2500.000
$130 | X-axis maximum travel | 400.000
$131 | Y-axis maximum travel | 430.000
$132 | Z-axis maximum travel | 1.000

Any advice?

thecuress, 2020-05-29:
OK ....IF use button and the laser work,,here is something wrong when you upload the file....put the file you are using here.....
p.s:change the line $31=0

WATSON-DOG, 2020-05-29:
yes have done no change

thecuress, 2020-05-29:
link the file you are using here.....

Dartanyen10, 2020-06-01:
I am having the same problem it was working fine for about a week and a half and now it will not burn but it will still run through the cycle like it is doing it. I can't figure out what has happen to the system or burn which ever one it is

closed 2020-12-16 by arkypita