arkypita/LaserGRBL #1024: Help

ironhorse2959, 2020-05-28:
I don't know a thing about lasers. I had ordered a cheep Super Carver off the internet. It came with software that worked for 1 day then vanished from my computer. Now I have the Super Carver but no software. I have no idea what the machine is programmed to run but hoping it may be grbl? Can anyone help me with this issue?

thecuress, 2020-05-28:
TRY to install lasergrbl

Schwarzi07, 2020-05-28:
Could you please provide more Information absolut the engraver.

thecuress, 2020-05-28:
i download the softaware....i try and I tell you

thecuress, 2020-05-28:
try this link   https://smartcarve.software.informer.com/download/

ironhorse2959, 2020-05-28:
Schwarzi07... It's a SuperCarver 100W laser. I  did install lasergrbl and on com 3 baud 9600 it homes but wont connect. I tried benbox but none of the .hex files work. Totally new to this and totally confused?

ironhorse2959, 2020-05-28:
thecuress... I tried to download smartcarv but it wants do download an .rar

thecuress, 2020-05-28:
you have not winrar for decompres the file?

StuartB4, 2020-05-29:
Change the Baudrate to 57600.

closed 2020-12-16 by arkypita