arkypita/LaserGRBL #1025: Laser not working at full power when engraving

reborneuk, 2020-05-28:
Hi, I'm using the hardware from my old Ender 2 3D printer, with an Uno and CNC Shield V3.  The laser is a 2.5w which came with a controller board LY-BC-0018 Rev 1-A. The controller socket from the CNC Shield is 2-wire not 3.  On power up the laser comes on for about 1 sec at full power.
I'm using Gcode sender.  Created a simple star shape to test - it all works fine, goes to the start point and the laser comes on, tracks the shape and stops as it should.
The problem is the laser only comes on at a low power and doesn't burn.  I have it connected to Z+ and GND on the shield. If I disconnect it from the shield during the run, the laser goes to full power.
Any help would be appreciated, thanks in advance.

thecuress, 2020-05-29:
it's like this ?  https://www.ebay.com/p/16016762146

reborneuk, 2020-06-02:
> it's like this ? https://www.ebay.com/p/16016762146

That's the one, yes. Cheers

StuartB4, 2020-06-02:
Does the laser driver board have the TTL/Analogue switch, if it has it should be switched to TTL.
Also can you post a list of your grbl configuration settings.

reborneuk, 2020-06-02:
Thanks Stuart. I can't see a switch on the board, but the socket for the trigger wire from the CNC shield is marked TTL.  I'll get to the computer later and get the current settings. Cheers.

StuartB4, 2020-06-02:
Ok.

reborneuk, 2020-06-03:
Hi, settings as requested;

**** Connected to COM7 @ 115200 baud ****
Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 1    (dir port invert mask:00000001)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 50.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 5.000    (x, step/mm)
$101 = 5.000    (y, step/mm)
$102 = 500.000    (z, step/mm)
$110 = 120.000    (x max rate, mm/min)
$111 = 120.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 5.000    (x accel, mm/sec^2)
$121 = 5.000    (y accel, mm/sec^2)
$122 = 500.000    (z accel, mm/sec^2)
$130 = 125.000    (x max travel, mm)
$131 = 250.000    (y max travel, mm)
$132 = 500.000    (z max travel, mm)

I also found an article on Instructables with someone using a similar setup - they had a 1k resistor between the +ve and -ve of the TTL output from the CNC shield.  I've also tried this, with no success (output from the Z- pins).

Many thanks!

gmmanonymus111, 2020-06-03:
Grbl 0.9 uses the old pinout, so you shouldn't need to switch TTL/Z pins on the shield. If you can do that, update GRBL to 1.1 (this way you can use lasermode, but as you did before, TTL signal should be read from Z- or Z+ pins), otherwise try using the TTL pin for TTL signal.

StuartB4, 2020-06-03:
As gmmanonymus111 said. You should upgrade grbl to 1.1. Then with that shield and uno you can leave the wiring as it is. With the PWM/TTL wire from the laser/driver connected to the Z + or Z- pin. 

You can do it in LaserGRBL on the Tools menu.
Here are the steps you need to do. But instead of selecting Nano, select UNO from the drop down list.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

reborneuk, 2020-06-03:
Many thanks, I'll give this a shot!

reborneuk, 2020-06-03:
Really big thanks guys, those upgrade instructions were faultless. I still hit a hurdle with it though, as it went a little brighter, but still not enough to burn.
However, I cast my mind back a couple of days to something I found which the board wouldn't accept, and retried it under the new version and it worked! $30, $31 and $32 settings sorted, I'm burning bright 😊
Now I just need to sort my steps and dimensions!
Again, big thank you!

StuartB4, 2020-06-03:
Thats good to know.
You can set the steps on this website with the online calibrator. 
https://jscalc.io/calc/tHjfIiKMKjDF42Hg?fbclid=IwAR219p6S0qWzVAsY8aLC8nyQaXaMpfa-mSTNCIXA7TAMPa3VzrGoCDcimsI

This is what you do. 

[image]

PaulC2577, 2020-07-27:
Hi. Hope you can help. It would appear a complete novice. I purchased the same kit, but can't work out the connections to the board. There doesn't seem to be any clear guidance on the board. The outs seem to be ins etc. I am going to fit it to a homemade CNC router using the Arduino uno, which I have had no problems.

StuartB4, 2020-07-27:
@PaulC2577

I just replied to your other post here: https://github.com/arkypita/LaserGRBL/issues/1083

closed 2020-12-16 by arkypita