arkypita/LaserGRBL #1026: spot removal not being applied when create target image

alanminner, 2020-05-29:
I am trying to set up engraving using vector but the spot removal setting is not being applied.  I am using version v3.3.1

gmmanonymus111, 2020-06-03:
+1
LaserGRBL shows the preview right, but does not actually remove anything.
[image]
[image]

arkypita, 2020-06-08:
Hi all
There is one thing to know when using raster import.
All the parameters taken have an effect on a preview image which is usually much smaller than that which is usually processed in the final phase. This is for performance issues in the preview generation.

This causes that the result could be very different from what you will see in the final image.

In a shrunken image, defects and "spots" are also shrunk, so a low threshold can easily remove them. When the image is instead treated at full resolution (or even sized up) the same threshold has no effect because the spots are much larger.

arkypita, 2020-06-08:
You can also see effect of this if you maximize/reduce the size of raster import form.

This is difficult to change, even accepting a longer preview design time, because the final size the user wants is not known in advance.

a solution could be to always apply vectorization to the original image (neither enlarged, nor reduced) so as to have a greater reproducibility of the result, both in the preview phase and in the marking phase on different sizes.

However, this changes may not be entirely appreciated as it is an interesting thing that the drawing is "simplified" and "cleaned" automatically when printed in small size, and that it is enriched with details when reproduced in large dimensions.

closed 2020-12-16 by arkypita