arkypita/LaserGRBL #1028: Ooops! Something went wrong

kurtpart, 2020-05-30:
LaserGrbl v3.4.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Professional, 10.0.18363.0 (7)

TypeOf exception  [System.ObjectDisposedException]
Exception message [Güvenli tanıtıcı kapatıldı]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   konum: System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   konum: Microsoft.Win32.UnsafeNativeMethods.WaitCommEvent(SafeFileHandle hFile, Int32* lpEvtMask, NativeOverlapped* lpOverlapped)
   konum: System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   konum: System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   konum: System.Threading.ThreadHelper.ThreadStart()

arkypita, 2020-06-01:
Same issue of #990 

I am investigating the issue. Already tried apply a fix in v3.4.0 but, as for your report says, it seem not sufficient. 

Let me know if the problem is sporadic or if it shows up every time you use LaserGRBL and if you are aware of any correlation with actions that take place when you had the issue.

arkypita, 2021-01-01:
Hi
I write a new SerialUSB wrapper to comunicate with the engraver, that replace microsoft implementation that has some bug like the one reported here. 

If you want you can test by download from here: https://github.com/arkypita/LaserGRBL/releases/download/v3.8.10/install.exe and enable UsbSerial2 connection protocol (before connect!).

[image]