arkypita/LaserGRBL #1030: Installer flagged by Windows Defender

kodosexe, 2020-06-02:
Hey,

the Windows installer is being flagged by Windows Defender SmartScreen as a potential problem when you try to launch it, bacause it is an "unknown app".
It says there is no signature available and therefore it doesn't trust the program.

The screenshot is attached, but it's in German.
[image]

arkypita, 2020-06-02:
Known issue of false positive from AV

http://lasergrbl.com/faq/
#235
#70

I'm sorry for that, but it's something I cannot deal with it.

arkypita, 2020-06-02:
Signing software require to pay for a signing certificate. 

As free software developed in my spare time I decide, for now, to not pay for this service.

closed 2020-06-02 by arkypita