arkypita/LaserGRBL #1031: Can not connect

dewaynemeats, 2020-06-03:
I am using com 4
I click on connect and will not connect. says not connected
Help
Dewayne

Maxoch, 2020-06-03:
Проверте целостность кабеля, проверте установлен ли драйвер. К другому ПО пробовали подключаться?

gmmanonymus111, 2020-06-03:
Are you sure your machine is connected to COM4? Does Baudrate set to the correct value?
Does your machine uses GRBL?
Can the USB cable transfer data, or is it just a charging cable?

gmmanonymus111, 2020-06-03:
Baud rate is the 'communication speed' between your PC and Arduino
You can set it here
[image]
It is usually `9600` or `115200`
Could you tell us more about your machine? Type, GRBL version (if known) etc?

closed 2020-12-16 by arkypita