arkypita/LaserGRBL #1034: Vectorize: fill shifted

IgnasGi, 2020-06-03:
Bug found. 
The fill is shifted relative to the contour in Vectorize! tool.  Small images do not look good.
Shift logic is the same for all images, as I tested.
For horizontal fill: it's shifted 3/4 line down.
For vertical fill: it's shifted 3/4 line left.
Diagonal fill is OK! But it's placed denser. It's obvious in picture below: compare vertical and diagonal fill.
Adaptive fill option creates even more problems: it shifts fill unevenly! By one line only but in all fill mode, diagonal too!
The only way to fill good is fill diagonal with Adaptive option turned Off. But in this case fill is dense than expected ;)
Thanks for a nice program anyway! :)
[image]
[image]

arkypita, 2020-06-04:
Yes, it seems to be a bug.

I have always known about this but I have never investigated the problem because with a sufficiently high number of lines / mm (from 5 upwards) the defect is limited.

I am not sure what the cause may be or if a solution is possible because the vectorization + filling process is complex and goes through several steps.

[image]

The image is first vectorized, this produces paths that can be virtually in any position of the space, without any alignment between them and with the scan lines.

[image]

The vectorized image is first transformed into a black filled image, with a small internal offset and then transformed again into a raster image with a few of smoothing applied on borders.

[image]

[image]

Finally, the raster image is processed by the "line2line" algorithm, the same used for grayscale photos.

[image]

It is quite inevitable that there is no perfect alignment between the result of vectorization and the scan lines of the filling processing. This is more evident if you think of halving the lines of the previous image, let's say keep one yes and one no.

[image]

Although this is inevitable, it seems that LaserGRBL also has an offset problem. Judging by the result, it seems that the filling always has a constant shift that could probably be adjusted.

[image]

As for the more density of the diagonal fill, this is simply justified by the fact that in the diagonal line, with the same pitch X and Y, the resulting line is spaced by the side/diagonal root ratio of sqrt(2)

arkypita, 2020-06-04:
As for the more density of the diagonal fill, this is simply justified by the fact that in the diagonal line, with the same pitch X and Y, the resulting line is spaced by the side/diagonal root ratio of sqrt(2)

[image]

IgnasGi, 2020-06-04:
Proposal: do not bother about cause of problem but simply move Fill pattern a little ;)
Shift 3/4 of the line pitch, by my observations.
About filling density you are not right. In a picture above density is not 10/5= 2 lines per cm but 10/3.54 = 2,8.. lines per cm.
Check Adaptive fill option also. There is ilustration when Adaptive is turned ON:
[image]

arkypita, 2020-06-04:
> Proposal: do not bother about cause of problem but simply move Fill pattern a little ;)
> Shift 3/4 of the line pitch, by my observations.

I haven't reviewed the code yet, but keep in mind that Line2Line work with the image scaled to a number of pixel equal to resolution (line/mm) * size, so an image of 1cm*1cm at 10lines/mm produce a bitmap of 100x100 pixels where each line of pixels correspond to one line of laser scan.

When you work with pixels there is no 3/4 pixel.
Either move one pixel, or move two, or move zero.

You must also keep in mind that it is mandatory to respect a positioning of the lines that is an exact multiple of the set step. This is because the user may want to set a specific step as a multiple of his hardware resolution to avoid ripple/rifling effects #30

This could be done with resolution helper form.

[image]

So if the user asked to make a line every millimeter because its laser is able to step exacly at multiple of 1 mm, it is mandatory that I generate the lines at 0, 1, 2, 3, 4 ... etc millimeters, and that is exactly what LaserGRBL do.

[image]

[image]

I think now it's more obvious to you why it's not such a simple thing to change!

IgnasGi, 2020-06-04:
Thanks for a detailed explanation! But not sure that I understood everything. I'm novice in CNC.
I just want to inform ;)
I'm not speaking about very small images and pixels. The shift effect occur in all images. 
In this case, line spacing is 1mm. Fill is shifted 3/4 mm to the left.
Sorry for oftopic ... Can you help me please..
When I press Left button laser move to RIGHT!  How to fix it?  
The only way move to right is manually enter "G53 X-55"? :)
Cant set starting coordinate. Spend all day investigating.. HELP!
[image]

arkypita, 2020-06-04:
Axis direction is configured via $3
Read this: http://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

arkypita, 2020-06-04:
Both left and right move to the right?
Are you using continuos jog or not?

[image]

>But How to set/reset working area (coordinate 0,0)?

If you click on the globe with pin icon the current position is set as zero using G92 command.

Clear G92 shift is only possible
with a reset
or, moving back to previous position and press globe button again
or, perform homing if your hardware has limit switch

https://www.cnczone.com/forums/g-code-programing/162349-g92-clear-shift-values.html

IgnasGi, 2020-06-04:
Thanks! Now it's clear. Turned Off Continuos Jog.

arkypita, 2020-12-27:
Hi! I have made a small changes to try compensate the shift effect.
This need to be test, please download v3.8.8 and let me know 

https://github.com/arkypita/LaserGRBL/releases/tag/v3.8.8

yerpj, 2020-12-29:
See #1218 for some test report

closed 2021-02-07 by arkypita

IgnasGi, 2021-02-08:
Thanks A LOT :)

Ignas