arkypita/LaserGRBL #1036: Quality improvement in M4/PWM mode

IgnasGi, 2020-06-05:
Have a feature request ;)
Laser power regulation with PWM is not ideal. 
Grbl developer warns: "dynamic power mode may not work for all situations".
Especially it's vissible where laser accelerate - at the edges of the filling (picture).
When cutting or engraving wood it's not a problem, but in some subtle materials it's very visible, inappropriate.
PROPOSAL
Fixing this problem is easy: just add more extra space for laser to travel at edges.  5-10 mm ant the beginning and the end of the filling line is enough for motor/laser to gain suitable working speed. 
I propose to add one more parameter in Raster image import menu. Like:
**Filling overlap/bleed [mm]** - extra space to travel at the line ends in PWM mode
I hope this good idea will be implemented soon ;)  I look forward to it. 
This will very improve engraving quality in some cases.
[image]

arkypita, 2020-06-05:
I know the problem. It is alreay in development roadmap #64 but it is not so easy to implement due to the conflict of this feature with two other behaviour

1) the optimization that LaserGRBL apply that try to remove all empty movement (if I add extra space in the first phase of gcode generation the optimization will remove it in a second step)

2) the conflict of this feature with "phisical limit" of the engraver (if one want to engrave at 0,0 it could not have the extra space for travel)

arkypita, 2020-06-05:
I see some user solving this problem adding a very dim light background (or border) to their image that produce a light value for laser bright that does not burn at all.

I know it's not a great solution (it would be nicer if the software did it directly) but it's a good temporary compromise.

IgnasGi, 2020-06-05:
Thanks, good idea. It works :) But more work added.
Other solution will be add some PWM correction value (adjustable), to lower laser power at very low speed. But it's very laser dependent.
[image]

arkypita, 2020-06-05:
This difference is so significant in the results that it really motivates the development of this feature.
However it is not very simple to implement, otherwise I would have already done!

AxelGUILLAUMET, 2020-06-07:
Same problem there! Is there not an option into GRBL firmware to do that like curve parameters of acceleration/PWM out?

[](https://github.com/arkypita/LaserGRBL/issues/1044)

[](https://github.com/grbl/grbl/issues/1637)

Regards...

IgnasGi, 2020-06-10:
I wrote script for image editor  (Affinity photo)  that creates ghost duplicates.
LaserGRBL could automatically convert source image to this:
[image]