arkypita/LaserGRBL #1038: Laser on during cooldown

frankvdh, 2020-06-05:
I have the cooldown enabled... 5 minutes on, 1 minute off. Twice I have had the machine stop after 5 minutes, with the laser still on.
Not only does this damage the work, but possibly also the laser.

frankvdh, 2020-06-05:
Arrgh! Dummy. I didn't laser mode ($32) enabled

closed 2020-06-05 by frankvdh

StuartB4, 2020-06-05:
There is a warning in the settings where you enable it. 
If you have GRBL 1.1 you need to enable Laser mode in grbl configuration on the Grbl menu in LaserGRBL. Set the value of $32 to 1 and set the value of $30 to 255, then click Write and close.
If you have an older version of grbl you should upgrade it to 1.1

frankvdh, 2020-06-06:
Yeah, thanks. I saw that after I posted the ticket. I've closed the ticket
now.