arkypita/LaserGRBL #1040: Simple materials database

frankvdh, 2020-06-05:
Currently, I am spending a bit of time figuring out the right speed/power settings for cutting different materials (e.g. white card 300gsm or 5mm thick foamboard), and recording these in an Excel spreadsheet for future reference.

It would be convenient if this could be integrated into LaserGrbl... a combo and button or similar to allow me to save the current settings against a description of the material/process (e.g. "cut white card 300gsm", F400, S1000, 1 pass). I suggest these could be stored in a CSV for ease of loading/parsing/saving.

arkypita, 2020-06-06:
Really? Don't you know that latest version has this feature?

frankvdh, 2020-06-06:
No, I didn't know. I have auto-update enabled, but it didn't update, so didn't realise there was a newer version. I've downloaded v3.5.0 and it's exactly what I was thinking of. Thank you. :)

closed 2020-06-06 by frankvdh

arkypita, 2020-06-08:
> I have auto-update enabled, but it didn't update, so didn't realise there was a newer version.

Can you tell me what version do you have before? With "? - Open session log" you can browse all session log, where you can find info about LaserGRBL version recorded at each execution.

I noticed that on some PCs the auto-update function does not work correctly (or has stopped working) probably due to some windows update or due to the intervention of some firewall.

I made some changes in latest version which should solve the issue.

You can test if update is working for you by click on "? - Check Now!" menu.

If it say "you have the most updated version" it's all ok, if it say something like "Cannot check for new version" than auto-update does not work for you.

Let me know.

frankvdh, 2020-06-12:
I had v3.3.0. Since then I upgraded to v3.5.0, and when I started it up
this morning it automatically asked me if I wanted to upgrade to
v3.5.2pre-release.

So the auto-update feature is now working perfectly. :)

Many thanks for your hard work.

Frank

clcweis, 2022-09-01:
Dear frankvdh or arkypita:
I need a hero.
Give a gal a fish: Frank, what are the cutting parameters for "black card 176GSM" according to LaserGrbl material database?

Teach a gal to fish:  I'm not a comp-sci wizard.  Most of my rental kids are, but they are too busy to help me.  How do I get the LaserGrbl material database that helps with card stock and not just Kraft paper?  The former senior software engineer at Carvana, soon to be senior software engineer at OpenComp looked at issues/1040 on my behalf and sent me to https://lasergrbl.com/download/ to get the latest version.  But that's what I had already.  I reinstalled it anyway and still it gives me only engraving parameters for Kraft paper.

I have fed, comforted, and done laundry for so many CS majors in my university town.  They are now too busy making Tesla money to help me with this, which ironically will be used on personalized presents for them.  Have mercy on me?