arkypita/LaserGRBL #1041: Laser no longer burns any material (was working before latest update)

Actionslats, 2020-06-06:
Hi All. im having an issue that my 2.5W laser no longer burns. it produces the laser (blue in colour). within the 'config' window on lasergrbl i had changed the speed setting to match the laser output (2500). this was before the update. None of the settings or config appear to have changed, and im now at a loss as to what to do. Any and all ideas welcome. TIA

Actionslats, 2020-06-06:
the laser is also as focused as it was when it was working too

gmmanonymus111, 2020-06-06:
Since updating LaserGRBL shouldn't cause any problems like that, make sure your laser isn't dead.
Or, if you think the update broke something, you can downgrade by downloading an older version from [GitHub releases](https://github.com/arkypita/LaserGRBL/releases)

Actionslats, 2020-07-21:
I have now tried everything mentioned and more. Iv bought a new lense to eliminate that from the process. 
[image]
This is how the laser projects with no lense. Is this correct or is the diode knackered?

Actionslats, 2020-07-21:
Photo taken Parallel to laser housing/top rail

StuartB4, 2020-07-21:
That's pretty normal. Laser dots are actually rectangular. You just need to focus to the smallest you can get, it will never be a round dot though.

arkypita, 2020-07-21:
and this is why: [https://www.youtube.com/watch?v=NbHnxL44eXo](https://www.youtube.com/watch?v=NbHnxL44eXo)

StuartB4, 2020-07-21:
More people should see that video, would help them understand a bit more.

closed 2020-12-16 by arkypita