arkypita/LaserGRBL #1044: Edge co2 Laser engraving Issue in M4 laser mode

AxelGUILLAUMET, 2020-06-07:
Dear friends.

I'm trying to convert an old RF co2 laser cutter to a GRBL laser cutter, I solved everything but the problem I face now is the following:

[image]

When I slow down the machine this default is less.

What I think is happening is during the acceleration time, GRBL automatically adjust the PWM value while it's accelerating?

I looked into the source code at config.h

```
#define DEFAULT_SPINDLE_SPEED_OVERRIDE    100 // 100%. Don't change this value.
#define MAX_SPINDLE_SPEED_OVERRIDE        200 // Percent of programmed spindle speed (100-255). Usually 200%.
#define MIN_SPINDLE_SPEED_OVERRIDE         10 // Percent of programmed spindle speed (1-100). Usually 10%.
#define SPINDLE_OVERRIDE_COARSE_INCREMENT  10 // (1-99). Usually 10%.
#define SPINDLE_OVERRIDE_FINE_INCREMENT     1 // (1-99). Usually 1%.
```

It should exist a parameter to parameter this acceleration/pwm factor that no?

I use LightBurn.

Here is my GRBL configuration:

```
$0=3
$1=0
$2=7
$3=0
$4=1
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=3
$24=20000.000
$25=20000.000
$26=0
$27=1.500
$30=100
$31=0
$32=1
$100=58.000
$101=116.000
$102=500.000
$110=40000.000
$111=40000.000
$112=600.000
$120=7000.000
$121=5000.000
$122=400.000
$130=200.000
$131=300.000
$132=50.000
```

StuartB4, 2020-06-07:
I would set $30 to 255.

StuartB4, 2020-06-07:
Also it sholud be set the same in 2 places in lightburn. 

[image]

[image]

AxelGUILLAUMET, 2020-06-22:
Dear friend.

I checked my GRBL config and I have $30 to 255. And S-Max value in LightBurn to 255.

It doesn't fix...

EliteEng, 2020-06-22:
Hello @AxelGUILLAUMET I believe you need to uncomment and set pwm min value higher in config.h
`#define SPINDLE_PWM_MIN_VALUE 20`

The problem is CO2 lasers do not fire when they go under a certain pwm value.

AxelGUILLAUMET, 2020-06-22:
Tried to move the $31 GRBL parameter and it doesn't work...

[image]

What the difference between #define SPINDLE_PWM_MIN_VALUE 20 y $31?

Regards.

EliteEng, 2020-06-23:
$31 and SPINDLE_PWM_MIN_VALUE work very differently
This may be a bit confusing to explain but here I go.

$31 sets a min RPM, if you set your rpm (S Value) below this it will turn off the spindle/laser

SPINDLE_PWM_MIN_VALUE adds the value to the PWM output.

Looking at the code I believe neither of these will work as you expect, they don't work as I would expect them to work.

I would look at changing this line - spindle_control.c line 213 
from
`pwm_value = floor((rpm-settings.rpm_min)*pwm_gradient) + SPINDLE_PWM_MIN_VALUE;`
to
`pwm_value = floor((((SPINDLE_PWM_MAX_VALUE-rpm)/SPINDLE_PWM_MAX_VALUE)*SPINDLE_PWM_MIN_VALUE)+rpm);`

then set $31 to 1 and SPINDLE_PWM_MIN_VALUE to around 30-40 depending on when the laser fires.

This equation converts the 0-255 power levels to laser firing power to max laser power eg. 30-255

Please note this is untested but should do what you are after.
Also $31 has to stay at 1 or else the laser will not even try to fire at levels below $31

closed 2020-12-16 by arkypita