arkypita/LaserGRBL #1049: error

boseran, 2020-06-12:
LaserGrbl v3.5.0.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Professional, 10.0.18363.0 (7)

TypeOf exception  [System.ArgumentOutOfRangeException]
Exception message [InvalidArgument=Значение "0" недопустимо для "index".
Имя параметра: index]
Exception source  [System.Windows.Forms], thread [Main Thread]
Exception method  [System.Object get_Item(Int32)]

   ----------- stack trace -----------
   в System.Windows.Forms.ComboBox.ObjectCollection.get_Item(Int32 index)
   в System.Windows.Forms.ComboBox.get_Text()
   в System.Windows.Forms.ComboBox.WmReflectCommand(Message& m)
   в System.Windows.Forms.ComboBox.WndProc(Message& m)
   в System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2020-06-16:
Thanks for issue report. What are you doing when you had this error?
Any hint on how it could be reproduced?

boseran, 2020-06-26:
Hi!
Thank you for your reply.
I was doing the ordinary job: played «start» and after approximately 15 minutes the machine just stopped.  Meanwhile on the display of the computer the process was still active. The error and  «Send the report»  appeared after another 10-15 minutes. Then I pressed the button «Reset all the configuration» («lightning» button) and at the appeared window  p,,ressed « Continue from the shutdown point». RThe situation like this continued several times.  Even now sometimes this happens. 
Unfortunately I have no idea why that happened.
 
Best regards,
Sergey 

> 
> 
>Thanks for issue report. What are you doing when you had this error?
>Any hint on how it could be reproduced?

Maxoch, 2020-07-02:
У меня похожая ситуация. сле 45 минут работы я нажал "hold" чтобы лазер остыл. Через 10 минут я нажал "Resume", но машина не начала работать. Кнопка «Сбросить всю конфигурацию» («молния») тоже не помогла, помог только Hard Reset. Но восстановить координаты работы не удалось, так как я пока работаю без концевых выключателей - "Home is disable"

Maxoch, 2020-07-02:
LaserGrbl v3.5.3
Grbl v1.1h