arkypita/LaserGRBL #1051: Spot at beginning when engraving vector lines

MarkDowes, 2020-06-15:
Hi again.
So, I am continuing with my mission on cutting and marking paper and card using vector patterns.
Everything is going great but I having an issue when marking card, I am trying to burn some numbers on the surface of the card.
This is going great, except that at spots at the beginning of lines it cuts through a tiny little hole, especially on smaller more intricate marks, like letting or numbers.
I have played around for hours with junction deviation and acceleration and speed and power, and I cannot avoid these tiny little holes happening at the beginning of new lines and sometimes in sharp corners.
It seems to happening when the laser fires, either there is a pulse that is higher power or it might be that the motors are not moving yet or slowing down slightly and the laser is making its way through.
Other then those tiny moments of making it through I am getting perfect results.
Any advice on best settings to avoid this would be much appreciated.
I noticed in other software that there are settings to avoid this by lowing the power for X ms on corners and on fire, is there any way of doing this in LaserGRBL? If not I would love to see it added.

As always, thanks for your time and continued development of this software. Love the recent updates, SVG import works brilliantly now!

Cheers.

arkypita, 2020-06-16:
Did you use M4 or M3?

MarkDowes, 2020-06-16:
I just had to look it up. M3. I see from my quick google search that M4 might solve my problem. I will try this. Thank you loads!

closed 2020-12-16 by arkypita