arkypita/LaserGRBL #1057: Can't connect to Neje Master 2 with 7W Laser

jamfx, 2020-06-21:
Hi, as some others here, I have the Problem, that I can not connect LaserGRBL to my Neje Master 2 Laser Cutter. You can find mor about it here: http://wiki.nejetool.com/doku.php?id=nejelaser_master_2
The Neje-Software and the Android App can connect without any problem via COM5 (Of course via USB, but Software says COM5).
Any hints, any tests, any thin I could help, to get this thing working?
Kind Regards
Niko

gmmanonymus111, 2020-06-21:
It seems neje installed only the benbox software by default. You need to flash GRBL to be able to use LaserGRBL.
From their website:
>The software marked “√” is equipped with all functions.
NEJE firmware is good at efficient SCANNING mode and unique minimal Gcode mode. High quality, high efficiency, simple, safe, portable.
Atmel supports Benbox firmware and GRBL1.1f. **The default is to install Benbox firmware. If you want to use LaserGRBL software, you can download the GRBL uploader on the wiki.**

jamfx, 2020-06-21:
Hi, thanks for the reply. I don't want to loose any functionality. Will I be still able to use the NeJe-Software? Can this Laserboard run somehow like two Firmwares or do I have to use GRBL for then on?
Thanks!

gmmanonymus111, 2020-06-21:
That is also written there. The engraver uses 2 controllers, one with Neje FW, the other can be flashed with either Benbox or GRBL

jamfx, 2020-06-21:
Thanks. It worked for me, like you said. I just flushed the GRBL-Software, too! Great! Thanks for your quick reply and help!

closed 2020-06-21 by jamfx

gmmanonymus111, 2020-06-21:
I'm glad it worked out