arkypita/LaserGRBL #1059: Problem with OrturLasermaster 15W and LaserGrbl

MCSpider65, 2020-06-25:
Hey Community,

I hope you can help me. I bought a Ortur Lasermaster 15 W. It works fantastic on the first day, on second day (after restart the computer) I got a "Soft limit" Error.
I did a reset, new installing all drivers and software, but nothing works.
Hope somebody can help me, otherwise it was waste money and I put it away. I don´t know what to do.

I put some screenshots helping to show the problem.

[image]
[image]
[image]
[image]

arkypita, 2020-06-26:
The soft-limit error is generated by the hardware to prevent you from moving the laser beyond the physical limits of the axes.
Each command sent (jog or marking) is controlled according to the frame size configured in grbl configuration.

[image]

arkypita, 2020-06-26:
You should check travel size you set in $130-$131 and/or you can disable soft limit by set $20 to 0

closed 2020-12-16 by arkypita