arkypita/LaserGRBL #1060: Request for Insight or Help to Fix (Any Advice Appreciated)

Stardog2537, 2020-06-27:
Hello,

I'm a new hobbyist, and I'm having some issues with this software and the machine I recently purchased. I don't think it has a brand name, but that's beside the issue. So, the laser works alright, more or less, except for when attempting to use the lasergrbl vector option. When I send an image as vector to the laser, all isolated elements that do not touch the outline get printed over each other at the bottom of the engraving area. Anything that is part of the outline gets engraved as intended, but if there is a cut out area apart from the outline, like for eyes as an example, those would get burned down at the bottom of the print area. It's almost as if there are no Y-axis values being passed to the board. I went into an image editor to create a work around by making a pass through to those areas so they'd not be cut off from the outline. Doing that works, but isn't ideal. A lot of complex images are off limits to me until I get this sorted.

Thanks in advance.

StuartB4, 2020-06-27:
Can you post a picture of your grbl settings. Start LaserGRBL and click Connect.
Goto Grbl menu and click Grbl configuration. You should get a list of the settings.
Post it here. You may need to calibrate the steps/mm for the Y axis.

sinfocomp, 2020-06-27:
convert to curves

Stardog2537, 2020-06-27:
Hi, 

Thank you. The screen cap for it is here:
[image]
Interesting you should mention that. I didn't bring it up, but the measurements are off for both axes. The engraving actually is almost half the size of the actual settings when I run a job. For instance, one job I ran came out close to 2.9 inches when the mm settings should've resulted in a job that was almost 6 (maybe 5.8) inches. They do at least appear to be in the proper proportion for both directions.

Stardog2537, 2020-06-27:
Okay, thanks to your helpful post, I was able to calibrate the stepper motors to the proper measurement. Now, how do I convert to curves?

Stardog2537, 2020-06-27:
Now it's doing something else. Let me show you please:
[image]
When this figure is being run, the marker that shows the laser position actually goes to the correct places in the image as it does each part.

[image]
But when it actually executes at the machine, you can see the mouth part get placed at the bottom of the image. Strangely, this time through, it's placing the sections behind the knee about where they should go. Yesterday and the day before, it was placing those parts at the bottom around where the mouth parts are going. 

The misalignment issue has to do with the stepper motors I'm fairly certain. The outline misalignment is greatly corrected when I slow the speed way down.

Stardog2537, 2020-06-27:
Is this due to user error or software?

StuartB4, 2020-06-28:
Which format is the image you loaded, jpg, bmp, svg .....
Can you upload the origanal here for me to have a look at.

Stardog2537, 2020-06-28:
[image]
This is the file. I just take it into LaserGRBL and try to adjust settings to vectorize it. 

The second file below is the workaround I made. By connecting the interior spaces to the outline this way, I avoid the issue altogether.
[image]

Stardog2537, 2020-06-28:
The workaround is fine for this image, but for more advanced images, it seems to just dump all the interior outlines together at the bottom of the image. I don't have another example handy at the moment.

Stardog2537, 2020-06-28:
Oh, sorry, it's a PNG.

Stardog2537, 2020-06-28:
Strangely, watching the tool marker move in the laserGRBL window, it indeed appears to move up to the appropriate spot in the image to engrave the areas it actually puts at the bottom, and then the tool moves to the outline where it should continue from there. I wonder if I should post the g-code also.

StuartB4, 2020-06-28:
As far as the image goes it looks fine. Do you load the image in to lasergrbl as it is, or did you vectorize it in another app
like inkscape first.
I just did both the images you posted twice each and both burned in the correct place.
No editing or anything, just loaded as they are.  The largest one is only 60 mm high because I don't have a lot of material to work on at the minute.

[image]

Stardog2537, 2020-06-29:
I vectorize in LaserGRBL.

arkypita, 2020-06-29:
99% you have too high acceleration set in your grbl configuration.
Read this guide: http://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

Stardog2537, 2020-06-29:
First, thank you (all of you). I appreciate your help. I have reduced the acceleration, as well as the overall speed. Also, this helped me fix the measurement calibration.

I will try it again with the new settings this afternoon/evening and post the results back here. I cannot use my workshop right now.

closed 2020-12-16 by arkypita