arkypita/LaserGRBL #1062: Engraving problem with M3/M5?

gradella, 2020-06-27:
Hello,

I am trying to help a friend that has no clue about programming or anything inside the Sain Smart Genmitsu CNC Router 3018 PRO. He works with wood and would like to engrave some cool stuff. I am starting to learn about Arduino for fun and I said I could help him. Everything was messed-up because a lot of people tried to help him in the past two years but could not make it right. I would like to finish this job.

Now that I told you about this sad history, lets get to work:

I finally managed installing everything as said in the Manual and I have succesed by connecting the machine and my computer:

First the driver CH340SER (my PC)
Then the firmware upgrade 1.1 Xload (Machine)

And then I installed the Candle and Laser GRBL v3.5.3

Everything was nice till I try to engrave.

First problem was that the laser never turned off.
Find this post: https://github.com/arkypita/LaserGRBL/issues/521.... I changed $30-255,   $31-0 and $32-1.
It started to work the turn ON/OFF but seems like that it is switched.....That means the laser is turned on during gap movements (despite M5 is showed on G-code table) and turned off during engraving movements (despite M3 is showed on G-code table).

[image]
[image]

Now I am stucked without knowing what to do, and I really would like to give back the machine running "TIPTOP"

If someone know what to do, I am all yours.

How the laser looks like, could not find any document about model:
[image]

THANK YOU VERY MUCH!!!!!!!

StuartB4, 2020-06-28:
Do you have an offline controller connected as well as the usb lead, these machines don't work properly if both are connected.

Is the little switch on the laser driver board pressed in to enable PWM?

Do you have a seperate power supply (12volts and about 3-5 amps). If you do you
can connect it to the barrel connector on the laser driver board (top right). Then just use the white and black leads from the sainsmart controller board PWM and Ground pins to the PWM and Ground pins of the 3 pin connector on the laser driver board.
That way the laser runs on it's own power and should only switch on when it receives a signal from the PWM pin on the main controller, which would be M3 or M4 for on and M5 for off.

Are you sure the $30 and $32 settings were stored in grbl configuration settings. 
You can only change the settings when LaserGRBL is connected.

gradella, 2020-06-28:
Hello Stuart,

The offline controler is not connected.

I need to put the swicth on or the laser will not turn on at all.

About the power supply I need to buy one. But what do you think about take an adaptor and connect the red lead on the barrel connector as an option for a new power supply? should it works?...another question If I may (learning purpose)...what should be the difference in the circuit of the Laser board to works well with the barrel connector and not proper with the red wire?

The changes on $30 and $32 I made with the machine connected.

StuartB4, 2020-06-28:
This is all you need to do. Don't use the red wire from the 3 pin plug at all.

[image]

gradella, 2020-06-28:
Thank you very much,

Once I have the suplly I will test it and post here.

THANK YOUUU!!!!!!

StuartB4, 2020-06-28:
Ok. 👍🏻

ptcruiserx3, 2020-07-11:
hi does it really matter if you connect a 12v directly to the driver board via barrel jack as compared to using 12v from controller board to the driver board? whats the difference? advanatges? disadvanatges? i currently do not use the barrel jack 12v input..i just have 12v coming from controller 2pin laser connector to the dc input 2pin connector on the driver baord...and have had no issues for 2 years

StuartB4, 2020-07-11:
Should'nt make any difference, it just takes the strain off the main controller boards power if you use the laser's own seperate supply.

closed 2020-12-16 by arkypita