arkypita/LaserGRBL #1065: Graver du texte

Mickey013, 2020-07-04:
Bonjour,
Je débute avec ce logiciel, mais je ne trouve pas comment faire pour graver du texte (si c'est possible).
Serait-il possible de l'implémenter dans une version future ?
par exemple :
Une case de saisie de texte avec selection de la police avec toutes ses propriété.

Merci

gmmanonymus111, 2020-07-04:
I don't know if such feature will be implemented, but you could create an image with your own text & font, and import it to LaserGRBL. You can even crop it automatically that way.
If you need to do it again, it'll be easier. If it is a one time job, well, your idea could be better as it is faster.

Mickey013, 2020-07-04:
Yes, I can do like this (Or use an other software for this feature). it's just that I miss this functionality so that the software is perfect and that I only use this one. He could do the conversion by itself before load it as a standard bitmap and resume normal charging cycle (it's just an idea)