arkypita/LaserGRBL #1067: Material DB Is there an export and import function?

onmaker, 2020-07-05:
Does the material database have export and import functions?
We press import and nothing happens!
But don't see the export function?
[image]

onmaker, 2020-07-05:
In addition, we have to translate Chinese for this.
Don't know how to proceed?

arkypita, 2020-07-05:
I have not managed an import-export of the DB. The import button you see is provided in case I release a new version of the DB with new elements.

Also translation is not supported in material DB, sorry!

The file "UserMaterials.psh" with the current DB (included user additions) can be found here:
[image]

The file "StandardMaterials.psh" with the default DB distributed by me can be found in LaserGRBL installation folder:
[image]

Both are xml files that can be opened in notepad++ if you want to translate, but at the moment I have not managed the possibility of having different versions x different languages... maybe in the future.

kojiro18, 2020-08-06:
Hello, I translated the original lasergrbl material base into French. I would like to add new materials and share them with my groups using a file how can I do I am blocked thank you