arkypita/LaserGRBL #1068: added Traditional Chinese, which is a more complete new translation function.

onmaker, 2020-07-05:
I added Traditional Chinese, which is a more complete new translation function.
But I don’t know if my usage is correct, I have uploaded support and made a request.
Can you help me determine?#1066

arkypita, 2020-07-05:
hi! thanks for your translation.
give me time, i am a little busy this days

i'll check your submition asap and build a pre-release so you can check if it's all fine with your translation

onmaker, 2020-07-07:
Ok you have worked hard
Thanks for your help
If you have any questions about the translation, please tell me how to modify it

arkypita, 2020-07-09:
Just a simple question:

I know there are no good relations between China and Taiwan.
What is the right text and the right flags to use for the two (traditional chinese and simplified chinese) menu items? Something politically correct.

arkypita, 2020-07-09:
Could I use the same icon (china flag) for both?

arkypita, 2020-07-09:
Here is a pre-release https://github.com/arkypita/LaserGRBL/releases/tag/v3.5.4

onmaker, 2020-07-10:
[image]
Some of the displays here are limited, and may require a pull-out window or pull bar
--------------------
In addition, we will check later to verify how the program and translation work, and then update again.
In addition, there seems to be a place where Chinese has not been updated. I think this is because the area where he is located is the plugin area, so you cannot translate, or you have not put it? . as follows
[image]

onmaker, 2020-07-10:
> 我可以同時使用相同的圖標（中國國旗）嗎？

If you are distracted, you can put the flag of Hong Kong. They are in traditional Chinese.
This will not cause trouble.

arkypita, 2020-07-15:
I have seen that you also add a translation file for "StandardMaterials.psh" database.

Unfortunately for now I did not foresee the possibility of using translations in the material database because it uses a slightly different mechanism. I'll think about it, let's see what can be done.

arkypita, 2020-07-15:
As for the choice between traditional and simplified Chinese, I seem to have understood that the most correct language codes would be zh-Hans (simplified) and zh-Hant (traditional), which are language codes not linked to a specific territory.

zh-CN means "Simplified and using Mainland Chinese terms" and zh-TW means "Traditional and using Taiwanese terms"

https://en.wikipedia.org/wiki/Chinese_Wikipedia#Automatic_conversion_between_traditional_and_simplified_Chinese_characters

It therefore seems that hans / hant are more "all-inclusive" as a definition, while -TW or -HK are regional variants.

[image]

This makes you lose some automatisms that allow LaserGRBL to start automatically with the right language from the first run.
However this is a very small problem, because it is always possible to manually choose a language in LaserGRBL.