arkypita/LaserGRBL #1069: CNC Board with PWM laser will not work.

timtiernan, 2020-07-06:
Hi, I've tried everything I can think of but I'm really stuck! You're my last hope.

I have recently purchased a new CNC board (https://www.amazon.co.uk/dp/B07D211NKD/ref=cm_sw_r_cp_apa_i_FC5aFbWY4T8NA) and laser (https://www.amazon.co.uk/dp/B07VD47BMX/ref=cm_sw_r_cp_apa_i_cD5aFbKE41RB9) but they just will not talk to eachother.

I've got it wired as shown on the page for the board but for some reason when I send the following commands the laser doesn't switch on (PWM is enabled in settings).

[image]

I've tried it with and without the 12v external power to the driver board (adapter purchased separately (https://www.amazon.co.uk/dp/B017OMEW4Q/ref=cm_sw_r_cp_apa_i_RD5aFbARDEXH6)) but I'm not getting anything except the fans on the board and laser running when switched on (so I know it's getting power). I've also tried it with the jumper set to both laser and spindle with no effect. 

I don't believe that the board is faulty as it has s responding to LaserGRBL, just not doing anything on the other end.

I have had CNC boards and lasers work in the past but for some reason this one won't play ball.

Thanks in advance for any help offered.

[image]

[image]

[image]

Maxoch, 2020-07-07:
Попробуйте подключить сначала только +12-red, GND - black, +5 - yellow  нажмите на кнопку на драйвере. Если лазер запустится, значит он в порядке. тогда подключайте лазер как показано на картинки и быдет вам счастье))
[image]

gmmanonymus111, 2020-07-07:
You seem to send the power on command after enabling laser mode ($31). You need to be "moving" in laser mode to be able to turn the laser on. Use `G1F1M3S5`, or turn off laser mode (`$31=0`) and use `M3S5`.

timtiernan, 2020-07-08:
> Попробуйте подключить сначала только +12-red, GND - black, +5 - yellow нажмите на кнопку на драйвере. Если лазер запустится, значит он в порядке. тогда подключайте лазер как показано на картинки и быдет вам счастье))
> [image]

Hi, I only have one GND out on the driver board.

[image]

timtiernan, 2020-07-08:
> You seem to send the power on command after enabling laser mode ($31). You need to be "moving" in laser mode to be able to turn the laser on. Use `G1F1M3S5`, or turn off laser mode (`$31=0`) and use `M3S5`.
> EDIT:
> Maybe you need to replace the jumper from 'spn' to 'lsr' next to the middle driver. Also, are you sure your wiring is correct? The board you linked misses the 3-pin port where your laser is plugged.

Hi,

I've tried enumerable different wiring options to no avail. It does seem to be right, but for some reason it's just not picking up the PWM message.

Maxoch, 2020-07-09:
> Привет, у меня только один GND на плате водителя.

Я вижу, что GND только один, вам нудно переделать кабель. Я покупал лазер похожий на ваш и у меня в комплекте было два кабеля. Так вот я пользуюсь именно первым (картинка)
[image]

StuartB4, 2020-07-09:
Like this. 

[image]

[image]

closed 2020-12-16 by arkypita