arkypita/LaserGRBL #1071: LASERGRBL Laser will not turn on with manual M3 command using button

ptcruiserx3, 2020-07-11:
i have been using LASERGRBL for 2 years and it works fine for engraving images...however it will not turn laser on via manual command M3

i have tried M3 S255 ..still no laser on

i created a button so i can power on laser to see where my xy location is but it does not work..laser will not turn on

it shows the command in the window when i activate the button but laser does not turn on

laser works fine when doing gcode files in LASERGRBL

not being able to turn the laser on to see starting point makes engraving very difficult.

gmmanonymus111, 2020-07-11:
Try using `G1F1M3S255` or turn off laser mode.
When lasermode is enabled (`$32=1`) GRBL will only turn the laser on during movement.

ptcruiserx3, 2020-07-11:
Thanks ill try...however in the lasergrbl faq it does show sample button
for laser on as m3 sxx

I want to use for location purpose only so needs to not burn...its a 2w
laser what s value should i use?

ptcruiserx3, 2020-07-11:
And what you mean turn off laser mode?...my mschine is similar to
eleksmaker..its chinese knockoff...shouldnt laser mode be activated

gmmanonymus111, 2020-07-11:
Well, `S2` - `S10` should be enough, depending on the brightness you need

ptcruiserx3, 2020-07-11:
Ok thanks...i try
Its hard to rngrave without knowing start point so spotting is needed..ive
been eyeballing for 2yrs and its frustrsting

The laser on off button done per faq lasergrbl never worked

I hope this works

ptcruiserx3, 2020-07-11:
And i appreciated the fast response...

ptcruiserx3, 2020-07-11:
So for my laser button it would look like this

G1 F1 M3 S10

gmmanonymus111, 2020-07-11:
- Custom button one (disable lasermode):

Type: TwoStateButton
GCode:
```
$32=0
M3 S1
```
GCode2:
```
M5 S0
$32=1
```

- The second one ("fake moving"):

Type: TwoStateButton
GCode:
```
G1 F1 M3 S1 
```
GCode2:
```
G0 M5 S0
```
More buttons can be found [here](https://github.com/arkypita/LaserGRBL/issues/306)

ptcruiserx3, 2020-07-11:
U r awesome...dat worked

They need change faq button info on lasergrbl site

It burned at s10...so need to change lower ..unless s not working

ptcruiserx3, 2020-07-11:
It works great...as i change s value my pwm boltage is changing and laser
turning on with button and off so all is good now

Thank you

gmmanonymus111, 2020-07-11:
Glad I could help

ptcruiserx3, 2020-07-11:
S2 was giving me about .05v which seems about right

2/255 x 5v = .04v

ptcruiserx3, 2020-07-11:
See link

https://lasergrbl.com/usage/custom-buttons/

This is what made me scratch my head...

It say m3 Sxx

That no work..i think lasergrbl should revise that info

ptcruiserx3, 2020-07-12:
LASERGRBL is awesome program thank you

i have one suggestion though

when importing a svg file it does not allow you to resize the output
size  in lasergrbl for diff applications

it would be nice if it could be resized to whatever size needed for
each project...

when you import a jpg it allows output to be resized in lasergrbl
after it has been vectorized but for some reason it wont let you
resize a svg that is imported in

can this be done?

thanks

arkypita, 2020-07-13:
> i have been using LASERGRBL for 2 years and it works fine for engraving images...however it will not turn laser on via manual command M3
> 
> i have tried M3 S255 ..still no laser on

Just for knowledge: this is not an issue of LaserGRBL, it's an issue of GRBL!
Their [documentation about M3 in Laser Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) say:

[image]

So, in my opinion, M3 S10 should be enough to turn on laser in low power.

Ggmax4d, 2020-08-11:
HI folks
One thing I have is a 200 mHz scope DSO38 by Jyetech, enough 4 me. Then I put it between ground and Z pin (+ or -), that shows me if a square signal is sent to the laser. That helped me to detect some troubleshootings that, in fact came from most of the time bad electrical contacts.

ptcruiserx3, 2020-08-11:
I found issue...it was incorrect ttl connection...it fixed now

ptcruiserx3, 2020-08-11:
I ust ordered a cheap ebay k40 laser...i have heard for the money they
are ok...i think i have to use k40 whisperer on that one.

closed 2020-12-16 by arkypita