arkypita/LaserGRBL #1072: alarme s'affiche au bout de deux minutes

pinea81, 2020-07-11:
LaserGrbl v3.5.3.0
Grbl v1.1f
Wrapper: UsbSerial
Win 7, Home Premium, 6.1.7601.65536, Service Pack 1 (7)CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.8806 (QFE.050727-8800)

TypeOf exception  [System.ObjectDisposedException]
Exception message [Le handle sécurisé a été fermé]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   à System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   à Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   à System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Threading.ThreadHelper.ThreadStart()

arkypita, 2020-07-11:
Thanks for this information.
Unfortunately, this problem is driving me crazy. This is a problem in the microsoft code that occurs in very rare conditions and that I can't simulate, so I struggle to solve.

It has already been reported by other users, but almost nobody has been able to describe the situation / condition with which it occurs.

Is it a repetitive problem? Is there any procedure that makes it happen?

pinea81, 2020-07-12:
bonjour, j'ai préférer passer sur un ordinateur qui fonctionne avec Windows 10 , mais maintenant le cycle se coupe toujours au bout d'une minute et le message d'erreur est : unexpected reset
merci

closed 2020-12-16 by arkypita