arkypita/LaserGRBL #1074: Miglioramento asse Z

distebia, 2020-07-15:
Salve, nell'aggiornamento 3.1.3 è stato introdotto "aggiungi esportazione con intestazione / piè di pagina e passaggio multiplo" e ho pensato questo: supponiamo che eseguo un taglio e quindi ho la necessità di fare 2 passaggi affondando il secondo di 1mm. Perfetto usando
G91
G0 Z-1
G90
ma se voglio fare un altro taglio di un oggetto diverso, file diverso, mi ritrovo la Z fuori fuoco laser in quanto si trova a Z-1. Questo perchè finendo il lavoro non esegue il movimento inverso riportando la Z alla sua posizione iniziale.
Come potrei sistemare per fare in modo che dopo l'ultimo passaggio mi porta la Z in posizione iniziale?
Inoltre come potrei dire a GRBL di avere già la Z a +30mm al primo avvio? Non parlo di portare la Z a +30 ma di indicare che già si trova a+30mm. Grazie

gmmanonymus111, 2020-07-16:
For your laser to move to Z0 after it finished a job, you need to set the footer section in LaserGRBL.  `LaserGRBL` -> `Grbl` -> `GCode` tab, then as footer, type `G0 X0 Y0 Z0`. This will return all your axis' to 0.

For your second question, well... you need to 'manually' go to Z30, as I'm afraid I don't know an easy way to set position. The hard way would be using offsets, but that also requires setting it to 0 first.
I hope I understand I correctly, I used Google translate.

StuartB4, 2020-07-16:
I don't have a Z axis myself, but could'nt you just add a G0 Z30 movement after the G0 X0 Y0 Z0 in the footer.
That way after very job it would go to 0,0,0 and then move it Z 30.
Or does it not work like that.

gmmanonymus111, 2020-07-16:
@StuartB4 By google translate it seemed he does not want to move, as the laser is already there. He wants to tell it to GRBL as well, without moving.

distebia, 2020-07-16:
Sorry, but I did not understand

StuartB4, 2020-07-16:
That's what I understood as well, just thought that would be one way of having Z at 30. But then if he uses homing it would probably go back to 0 anyway as soon as he homes it. I can't think of any other way to of doing it.

distebia, 2020-07-17:
Se non esiste un modo su grbl di impostare una posizione fittizzia senza movimento potrei fare l'homing di Z in alto e poi scendere di quanto mi serve per portare l'asse a 30 dal basso verso l'alto. Tutto questo mi serve in quanto ho l'asse z disabilitata da firmware, non eseguo l'homing z, ma ho i limiti soft abilitati. La mia posizione di Z è fissa a 30 (manualmente), mio fuoco, ma se voglio fare passaggi multipli e scendere di 1 si attivano i limiti soft in quanto la Z per grbl è Z0. Quindi volevo dire a GRBL che già è a Z30 e posso scendere e salire di quanto voglio anche in funzione dello spessore del pezzo da lavorare.

closed 2020-12-16 by arkypita