arkypita/LaserGRBL #1075: LaserWeb/Octoprint

headly617, 2020-07-16:
Will also post this to octoprint site cause not sure where issue really comes from.  I am running octoprint server on a laptop for a 3d printer and using lasergrbl to connect to a engraver using esp8266 device and laseweb as per your website.  Tried telnet method but had constant disconnects.  Everything was working great until I noticed that if I stop or pause a print on octoprint the laser engraver will also stop or slow to a crawl.  I checked to make sure they were not somehow using same IP adress but they are not enything else to check?  Will try to recreate today to try and see what the error message was.

gmmanonymus111, 2020-07-16:
(Note: I used only Telnet, laserweb was a mess for me)
For me, esp8266 seemed to crash due to too fast status request from LaserGRBL. Try setting `Threading mode` to `Fast` or `Slow` (`LaserGRBL` -> `Grbl` -> `Settings` -> `Protocol` tab). Also, I switched to [ESP-Link](https://github.com/jeelabs/esp-link) as it was more stable than the standard WiFi-to-Serial bridges I tried (due to a bug in esp's wificlient), and can handle multiple connections at the same time.

arkypita, 2020-07-16:
Only few LaserGRBL user use WiFi to connect to their engraver, so I have very little feedback on it.

Personally I prefer use Telnet option, because it is more low level and give less overhead to the communication (each command exchanged via WebSocket is encapsulated in a web call).

However ESP8266-SerialWebSocket work by opening a websocket server on esp8266 side, that listen on port 81 ([see code](https://github.com/arkypita/ESP8266-SerialWebSocket/blob/c3c5adc440c38a5fad937d72011d8ee7be9ca3bb/websocketserver/websocketserver.ino#L22))

LaserGRBL will open a connection to the listener on port 81 which the TCP-IP system should move to a second random port specific for the connection (once opened websocket connection should be stable).

The octoprint server should not disturb this communication.
I would check for disturbance by firewall or antivirus instead.

arkypita, 2020-07-16:
Only few LaserGRBL user use WiFi to connect to their engraver, so I have very little feedback on it.

Personally I prefer use Telnet option, because it is more low level and give less overhead to the communication (each command exchanged via WebSocket is encapsulated in a web call).

However ESP8266-SerialWebSocket work by opening a websocket server on esp8266 side, that listen on port 81 ([see code](https://github.com/arkypita/ESP8266-SerialWebSocket/blob/c3c5adc440c38a5fad937d72011d8ee7be9ca3bb/websocketserver/websocketserver.ino#L22))

headly617, 2020-07-16:
When I was trying to get telnet working think I must have messed with some router settings.  Made 5 tests and now both seem to be working without interference.  Thanks for the help.

closed 2020-08-03 by arkypita