arkypita/LaserGRBL #1076: Feature request: M106 and M107 gcode commands for laser controll

ryanaukes, 2020-07-19:
Hello, I am using a laser which is controlled by the fan pwm signal, so I have to use the M106 P0 S??? command to turn the laser on and M107 P0 command to turn the laser off. Is it possible that this command gets added to the dropdown list with commands to turn the laser on and off? I really like the LaserGRBL software, but I cannot use it because I use different commands to control the laser. Thank you very much!

gmmanonymus111, 2020-07-19:
It is already in #64 AFAIK

>  - [ ]  Customized code for laser on-off

ryanaukes, 2020-07-19:
> It is already in #64 AFAIK
> 
> 
> 
> >  - [ ]  Customized code for laser on-off

I downloaded the latest version yesterday, but did nit find it in the dropdown list. Where can I download a version that does have it? Sorry, I dont have a lot of experience with github. Thanks again!

closed 2020-08-03 by arkypita