arkypita/LaserGRBL #1078: Engraving stops at the middle. Any advice is appreciated

khjung1992, 2020-07-24:
Hello,

I am a noob in this cnc laser world. I am starting to like this more than I though, indeed I enjoy reading this issues and learning from them. 

I build one mini simple engraver. I've been facing many issues, and resolving many of them, but this one I faced, I've got no idea. 
The laser engraver stops at the middle of the run.  The laserGRBL makes a beep sounds (pc) then it just stops running and not running any commands. I have to unplug the UNO and disconnect and reconnect.  But this happens again at every run.
It doesn't even show little window saying 'Problem detected' (I had this issue before, but after I changed my poor UNO to better one,  it seems like it doesn't happen again?) Please any advice would be great. PC makes a beep sound and engraving stops, it almost look like it freeze. 

Here is more detail about my printer.
I used 250mw laser with CNC shield v3 with UNO. Two stepper motor from DVD driver. 
[image]
You can see all the engraving stops at the middle.
[image]

[image]
When it freeze, It looks like this. I have 1.1f. 
I have tried all of buffer, sync, and repeatonerror with quiet setting, but no luck.
Thank you!

arkypita, 2020-07-24:
Looks a lot like my first engraver!

arkypita, 2020-07-24:
First of all check that the values ​​$100, $101, $102 are not too high. If you used a very strong microstepping they could be out of the range managed by arduino.

[image]

Next you have to try to discriminate what the problem is located.

1) disconnect arduino from the motors board, connect the naked and raw arduino to the PC, powered by the USB port only, and try to send the job.
- If the sending hang, the problem is in the PC or in the Arduino board (go to point 2).
- If the sending is done without problems, the problem is to be found in the electrical noise given by the motors (although usually the small motors like those of the DVDs give very little of it) or in the power supplies. (go to point 3)

2) discriminate if the problem is in the PC, in the cable, or in the arduino board. Try sending with another PC, with another cable, or to another arduino. Once you understand where the problem is, replace the buggy piece.

3) perform the same test without the motor board, but only by attaching the Arduino power supply (test if the issue is in the power supply noise), then try re-connecting the motor board (test if is in motor board), finally re-connecting the motors themselves (test for motor noise).

Be care: unpower motor board each time you connect/disconnect the board itself from arduino or from motors.

StuartB4, 2020-07-24:
The steps per mm for those little stepper motors are usually set at 53.333 for the $100 and $101.

khjung1992, 2020-07-24:
> First of all check that the values ​​$100, $101, $102 are not too high. If you used a very strong microstepping they could be out of the range managed by arduino.
> 
> [image]
> 
> Next you have to try to discriminate what the problem is located.
> 
> 1. disconnect arduino from the motors board, connect the naked and raw arduino to the PC, powered by the USB port only, and try to send the job.
> 
> * If the sending hang, the problem is in the PC or in the Arduino board (go to point 2).
> * If the sending is done without problems, the problem is to be found in the electrical noise given by the motors (although usually the small motors like those of the DVDs give very little of it) or in the power supplies. (go to point 3)
> 
> 1. discriminate if the problem is in the PC, in the cable, or in the arduino board. Try sending with another PC, with another cable, or to another arduino. Once you understand where the problem is, replace the buggy piece.
> 2. perform the same test without the motor board, but only by attaching the Arduino power supply (test if the issue is in the power supply noise), then try re-connecting the motor board (test if is in motor board), finally re-connecting the motors themselves (test for motor noise).
> 
> Be care: unpower motor board each time you connect/disconnect the board itself from arduino or from motors.

Thank you for the fast and detail reply! I've just tested with bare UNO, UNO with the cnc shield, UNO with the CNC shield with motors, and motor drviers and lasers. Basically I've tested all the combinations that's possible, and it turned out the issue seems it's coming from laser. When I run a program with motors, I got no issue. The engraving finishes all the way without any issue when I hook up everything without a laser. But If I disconnect all the motors, and just connect the laser, then during the engraving, it freeze just like I experience before. So I guess the laser is making too much noise? Or my wiring might be bad?
Could you guys check the wiring if it is possible? 
 
[image]
[image]

The gate of drain of mosfet goes to laser black wire. gate goes to Z+ pin on the shield, and source goes to the ground of the shield. And There is registers. 
[image]
https://www.youtube.com/watch?v=5HvFPYaVtQs 
I used the same wiring methods from this video, but I plugged my pwm to Z+ instead of SpnEn. 

I saw some previous issues, the other used relay between laser and the CNC shield. Do I need to use relay or is mosfet still able to use?

Also, It seems very strange that my engraver seems to have much bigger steps than what it tells form pc. I used one of the stepper calibration excel, and I used 40 for $100 and $101, but the motor was moving all the way to the end of the lead screws. I changed to 4 instead of 40, and now it seems more close to actual value. Is this normal? or It seems pretty lower than any other people who used similar stepper motors. (I know they are all different but still seems odd) And the speed is quite too fast so it doesn't have enough time to engrave the ground. It burns, but very thin. Is there other way to reduce the speed? I have tried changing $0 to 3 and when I upload a image file to program, I even tried to put 10 for the speed. is this something controls the speed of X and Y axis? 
[image]
Thank you!

khjung1992, 2020-07-24:
> The steps per mm for those little stepper motors are usually set at 53.333 for the $100 and $101.

Thanks StuartB4. I have tried 40 for the both of them but they were still running to far. It seems more correct when I put 4, but the number seems too low. Is low number okay? Or maybe I am missing something? Is a4988 driver potentiometer matters on the distance of the movement? I've tried changing $0 to 3 and lower the speed when I generate the image, but still seems bit fast. Thanks!

arkypita, 2020-07-24:
> The engraving finishes all the way without any issue when I hook up everything without a laser. But If I disconnect all the motors, and just connect the laser, then during the engraving, it freeze just like I experience before. So I guess the laser is making too much noise? 

Laser diode does not make electrical noise, only motor does.

> Do I need to use relay or is mosfet still able to use?

Do not use relay. Relay does not switch on/off so fast, mosfet does.

> Or my wiring might be bad?

Could be.

> But If I disconnect all the motors, and just connect the laser, then during the engraving, it freeze

You have a track: continues to investigate in this direction. Try with mosfet but without laser diode.
Try put a LED instead of Laser Diode.

Maybe your power source cannot give enough current, so voltage drop and arduino reset.

arkypita, 2020-07-24:
mosfet wiring looks good to me, also original circuit design seem to be ok

arkypita, 2020-07-24:
what mosfet did you use in your design? i cannot read by the pictures

khjung1992, 2020-07-24:
> what mosfet did you use in your design? i cannot read by the pictures

I purchased this one
[image]

khjung1992, 2020-07-24:
> Laser diode does not make electrical noise, only motor does.
>
>
> Do not use relay. Relay does not switch on/off so fast, mosfet does.
>
>
> Could be.
>
>
> You have a track: continues to investigate in this [...]

Thank you for the detailed reply! It's good to know that mosfet is faster than relay and correct to use this case. 
That is good point on the power. I will try with a different power supply, and led if I can find one. 
I've notice one thing when I unplug the 12v power supply while laser is running, the laser is still on. is that mean laser is driven by only laptop power? 

I got one more quick question. 
[image]
My LaserGRBL gets locked quite often. Every time when I turn on, It is locked as well, that I need to click unlock. Is there a reason for locking it by itself?

khjung1992, 2020-07-24:
> Laser diode does not make electrical noise, only motor does.
>
>
> Do not use relay. Relay does not switch on/off so fast, mosfet does.
>
>
> Could be.
>
>
> You have a track: continues to investigate in this [...]

Is that mean possibly my 12V power supply does not have sufficient current? Now I am looking at the different power supply with different amps for the same 12 voltage.

arkypita, 2020-07-24:
> My LaserGRBL gets locked quite often. Every time when I turn on,

Disable homing in $$ config
Also read some documentation https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbls-xval-settings-and-what-they-mean

khjung1992, 2020-07-26:
> > My LaserGRBL gets locked quite often. Every time when I turn on,
> 
> Disable homing in $$ config
> Also read some documentation https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbls-xval-settings-and-what-they-mean

Thank you for all the helps! 
I got it working over the weekend. 
You were right that the laser was drawing too much current from the board, and I was able to fix the problem by having additional exclusive power supply just for the laser. It took me little bit to realize to sync the ground of new supplier to the board, but it is working now. Thank you again for the help. I'm going to try build one more unit with bigger and more sturdy gantry and laser.

closed 2020-08-03 by arkypita