arkypita/LaserGRBL #1080: Hotkey Shortcuts

Focus3, 2020-07-26:
Good day,

i'm sorry to bother everyone with this, because it's not really much of an issue. I have been assigning Hotkeys which i use through my CNC pendant, and i wanted to assign two hotkeys to the "Speed". One to increase and one to decrease.

Is there a hotkey option for this? Because i can't find it in the Hotkeys window.

I have attached a picture.

[image]

thank you foryour help in advance.

gmmanonymus111, 2020-07-30:
I don't think that kind of Hotkey is implemented yet. You can only edit step size, or, G1 and G0 Multipliers

arkypita, 2020-08-04:
added: https://github.com/arkypita/LaserGRBL/releases/tag/v3.7.1

closed 2020-08-04 by arkypita