arkypita/LaserGRBL #1082: Vector image doubling in size with no option to adjust to bed size

rene0207, 2020-07-27:
[image]

I have been trying to bring in a vector image into the software but it does not allow me to adjust the size to match the laser engraver bed size. I created the file in illustrator using the bed size as the image size extents, so I know the file was correct originally, but doubles when imported into the Laser GRBL program. When I bring in a raster image it gives me this option. Is there an add on I have missed?

gmmanonymus111, 2020-07-30:
If you created it in Adobe Illustrator, it might be in a different DPI format (72 if I remember correctly), while the "standard" should be 92 DPI. If nothing works, you should try [Inkscape](https://inkscape.org/), at least for re-sizing your vector.
SVG Vector files have the engraving size embedded in them, so there is no need and no way (currently) for LaserGRBL to modify SVG sizes.

arkypita, 2020-07-30:
@rene0207  dealing with SVG files is a pain
Different program implement SVG file format in different way, making it very difficult to manage in all its variability.
The SVG implementation of LaserGRBL is tested with Inkscape, which I use as a reference.

As @gmmanonymus111 suggest you could import your SVG file in inkscape, size and position as your needs, then save it again: 99% LaserGRBL will load it correctly now.

closed 2020-08-03 by arkypita