arkypita/LaserGRBL #1083: Can't work out connection to laser driver board

PaulC2577, 2020-07-27:
Hi, I'm using an Arduino Uno and CNC Shield V3 driving a homegrown router/engraver. Bought a 2.5w laser which came with a controller board LY-BC-0018 Rev 1-A. The text on controller sockets don't make sense to me as the in's seem to be out's and there is no guidance as to where the power, laser, title connect to. The only one that is reasonably clear is the fan. Any help to this complete idiot would be gratefully received.

StuartB4, 2020-07-27:
Hi Paul. Can you please post some images of the parts and wiring you are using, laser, laser board showing the connections and any text/markings on the board near the connection sockets.  It would be easier if we can see what you have. 
Thanks.

StuartB4, 2020-07-27:
If this is the laser driver then it should be connected like this.
[image]

StuartB4, 2020-07-27:
On the CNC shield any of the pins down the right hand side of the board are ground, so you can use The Z+ or Z- for the PWM signal and one of the ground pins next to it. So you can use a 2 pin connector. Like this. My CNC shield is a Version 3.51 so the connection for the PWM (Yellow and Black wires) come from the SpnEn pin and Ground instead of the Z+/- pins, but it's basically the same.
[image]

PaulC2577, 2020-07-27:
Thanks very much for that StuartB4. That's exactly the one. The ac threw me as I only thought it would be12v dc. Also the fan "in". 
I thought I would ride in on the 'laser not working at full power when engraving 
' post, and I think you have responded to that as well. 
Hopefully, if I haven't already fried the board, should be up and running tomorrow.

StuartB4, 2020-07-27:
Ok. Yeah AC or DC is fine. The big black component on the laser driver is a bridge rectifier. It has 4 pins, 2 for AC in and 2 for DC out. It converts any AC to DC,  but you can still apply DC without any problem because it will just go straight through the diodes inside.  
[image]

closed 2020-12-16 by arkypita