arkypita/LaserGRBL #1086: *Safety Issue* version 3.7 does not turn off laser upon program exit

BernieMurphy, 2020-08-01:
Manually turning on the laser (say for focus) and then exiting the program, leaves the laser turned on.

Please treat as high priority

arkypita, 2020-08-03:
Laser safety issues are responsibility of controller hardware and firmware, they are not the responsibility of the high level software.

PC and its software can crash, be killed by taskmanager, have implementation error, run out of power, disconnect from controller etc. There is no possibility for high-level software to handle all these possible situations ensuring the safety of operations.

A good security implementation is the one implemented by the Ortur brand engravers, who turn off the laser as soon as they recognize the non-communication situation.

arkypita, 2020-08-03:
In few words, this safety issue does not concern LaserGRBL but the hardware manufacturer.

closed 2020-08-03 by arkypita