arkypita/LaserGRBL #1087: Version 3.7.0 does not have slider bars for manual step size

BernieMurphy, 2020-08-01:
I though I saw a screen shot with this feature.  Did I miss something?

Great program for beginner!

Thank you

StuartB4, 2020-08-01:
I think you have Continuous jog enabled on the Jog Control tab in the Settings on the Grbl menu.

BernieMurphy, 2020-08-02:
Thank you for the prompt reply! 

Actually, I did have Continuous Jog off.  I turned it on and then off and then the slider bars appeared.  Looks like an uninitialized variable in the code?

Great program - works extremely well with CNC 3018 PRO with Woodpecker control board.

I will send a donation via PayPal.

Thank you!

Bernie Murphy, Ottawa, Canada

(most typos in this message have been introduced by highly sophisticated auto-correction software)

StuartB4, 2020-08-02:
That's good. Easy fix, I like the easy ones. :+1:

closed 2020-08-03 by arkypita