arkypita/LaserGRBL #1089: Разлика при гравиране на един и същ файл в различна версия на програмата, в размера също

Deimion33, 2020-08-05:
Здравейте, инсталирах новата версия на програмата. Има проблем с гравирането, файлове запазени в старата програма се гравират гладко, без ръбове. Всеки нов файл файл направен в новата версия дава тези ръбове.
Променя и зададения размер- Н63, го прави Н62, трябва да се напише Н64, за да стане Н63.
Ще се радвам да ми отговорите бързо какво да се направи или да отстраните проблема.
Благодаря предварително
[image]
[image]
[image]

[image]

gmmanonymus111, 2020-08-05:
For the sizing, it seems your image is not cropped properly. LaserGRBL's preview shows only the size of the engraving area. You see 63 instead of 64 because you have a white edge while importing. Try using the auto trip feature which will remove all unnecessary white parts of the image. It should be the second icon from the right at `Import Raster Image` screen

Deimion33, 2020-08-05:
Благодаря за бързия отговор, 
Размера на старата версия не се променяше с толкова много. Показах снимка на файл от старата програма и на новата програма - използваната снимка на логото и размера са еднакви, но изработката от лазера е различна. Проблема не е в изображението. Затова снимах готов продукт гравирани са двете лога едно след друго с една и съща снимка. Красивата версия е стар файл направен в старата версия, този с ръбове е от новата.

gmmanonymus111, 2020-08-05:
Well then, do you know the old version number? If there were a code change, this way it can be found/fixed, but I don't think it was modified.
Are you sure both the old and the new engravings are in the same settings? (lines/mm, line-to-line engraving and not dithering, laser on command etc)

Deimion33, 2020-08-05:
Да, току що правя гравирането. На едно и също дарво със еднакво зададени параметри- размер, мощност и т.н. Имам 2 запазени файла от старата версия и те се гравират гладко, всичко друго, което направя излиза с ръб.
Новата версия промени размера на 14 на 63, старата е 15 на 63. Всичко в новата версия няма гладък контур при мен, назъбено е.
Това е втората нова версия с която заменям първоначалния софтуер. За първи път имам проблем при гравиране.
Не знам кода, смених с новата, аз не съм толкова добра с програмирането. Извинете, че ви безпокоя, но не знаех към кой да се обърна.

gmmanonymus111, 2020-08-05:
One last thing, before I run out of ideas. While importing your image, did you use `Smooth (HQ Bicubic)` or `Sharp (NearestNeighbor)`? If  you were using Sharp, try it with Smooth.

arkypita, 2020-08-05:
There are no software changes in the new version that can justify a different result.

The differences are to be found in the parameters used to import the image, in particular:
Resize (smooth or sharp) and B/W

[image]

Deimion33, 2020-08-05:
Много благодаря! Настройките, който използвах в старата програма бяха Quality 3000 и Smooth , ( в запазените файлове беше 3000 също) когато промених в новата на Quality 5000 , печата прекрасно! 
Още веднъж ви БЛАГОДАРЯ от сърце! и на gmmanonymus111  :)

closed 2020-12-16 by arkypita