arkypita/LaserGRBL #1091: Can't open DFX files. I purchased a fan, that should blow the smoke away, but it isn't working.

czecht, 2020-08-05:
I'm using the V2.3 version, all works so far, but the DFX files. When I try to open a DFX file, it behaves like I did nothing. The logo is the same as when I started the app.
Do I need to convert it or open in some other program first and re-save it with a new name ... etc?

Also, when I installed the fan, that blows the smoke away, it isn't working. When I measured the two pins, I see only 5V, but all the fans are 12V.
Do I need to configure something first?

Thank you very much.
Tony

StuartB4, 2020-08-05:
Are you sure you are asking in the right place?

czecht, 2020-08-05:
I never done this before - where should I ask questions about configuring
the software?

Thank you very much.

Tony

StuartB4, 2020-08-05:
Here is fine for configuring the software, not sure about the fan problem though without more info on your machine.

You cant load DXF files in. They need to be raster images, svg vector files or Gcode files.  .SVG, .JPG, .PNG, .BMP or GCode files like .nc

StuartB4, 2020-08-05:
Are you using LaserGRBL V2.3 ? Thats a really old version from 2017.

ThomasSc, 2020-08-06:
It looks like he is talking about "VigoEngraverL7_v2.3".

StuartB4, 2020-08-06:
Yeah may be.

czecht, 2020-08-06:
> Are you using LaserGRBL V2.3 ? Thats a really old version from 2017.

Where can I get a newer version? When I google it, I was able to download version 10, BUT there isn't any install app at all. I'm new at this, so I struggle much more than I should, but you guys are helpful to me! 
Also, if you can point me to some video or "xxx for dummies...laser cutter..." it would help also.
Thank you very much to all of you, who are helping me.

czecht, 2020-08-06:
> Here is fine for configuring the software, not sure about the fan problem though without more info on your machine.
> 
> You cant load DXF files in. They need to be raster images, svg vector files or Gcode files. .SVG, .JPG, .PNG, .BMP or GCode files like .nc

Is there a program that I can use to see what my DXF files are, and also, if I can get some kind of converter to make the files I have, compatible with my software, I'll try to use it. 
I was under the expression that DXF are all the same, but it seems to me that they aren't!?
Thanks again for your help.

czecht, 2020-08-06:
> Are you using LaserGRBL V2.3 ? Thats a really old version from 2017.

That is what my software is telling me in the left upper corner.
I was looking for something like: "Check here for newer version...", but the software is very limited.
Is there a free program out there that would work better?
If I have to buy one, what is the software majority users, using the same laser cutter, are using?
Thank you very much again!

StuartB4, 2020-08-06:
You should update LaserGRBL. You can get it here:
https://github.com/arkypita/LaserGRBL/releases/tag/v3.7.1
Just click the install.exe to download it.

StuartB4, 2020-08-06:
`If I have to buy one, what is the software majority users, using the same laser cutter, are using?`

You have'nt told us what laser machine you have.

czecht, 2020-08-06:
> Here is fine for configuring the software, not sure about the fan problem though without more info on your machine.
> 
> You cant load DXF files in. They need to be raster images, svg vector files or Gcode files. .SVG, .JPG, .PNG, .BMP or GCode files like .nc

Usually there is something in SETUP, that turns stuff on/off but nothing I was able to find.
When I measured the output voltage on the 2 remaining pins that have no connector, it is showing 5V DC, but all my fans I have are 12V. 
I use VigoEngraverL7 v2.3
Thank you very much for your help.

czecht, 2020-08-06:
> Here is fine for configuring the software, not sure about the fan problem though without more info on your machine.
> 
> You cant load DXF files in. They need to be raster images, svg vector files or Gcode files. .SVG, .JPG, .PNG, .BMP or GCode files like .nc

I am able to open IMAGE that is JPG - I haven't tried any other extension you are mentioning above.
It has a button for Gcode, Image, DXF, Text, Gallery, Settings.
No problem using the JPG or the Text options.
I haven't tried the other extensions as of today.
Thank you very much for helping me!

StuartB4, 2020-08-06:
This is the Vigotec's website. http://www.vigotec.cn
It looks like version 2.3 is the latest.
But because you are here on the LaserGRBL site you should get LaserGRBL.
You will get more help if you are using that. It's also better software than the vigotec software.

czecht, 2020-08-06:
I'm so sorry!
I was using a different software all together.
I guess my frustration spilled so far from the truth, that I left all this
mess!
I need to delete all of my questions and find the appropriate support!
Thanks for all of your help!!!

czecht, 2020-08-06:
> Are you sure you are asking in the right place?
I'm so sorry!
I was using a different software all together.
I guess my frustration spilled so far from the truth, that I left all this mess!
I need to delete all of my questions and find the appropriate support!
Thanks for all of your help!!!
My software is called: VigoEngraverL7 v2.3

closed 2020-08-06 by czecht

StuartB4, 2020-08-06:
No problem. If you do decide to use LaserGRBL in the future, this is the place to come to.
Good luck with it. : )