arkypita/LaserGRBL #1092: Pb Material

kojiro18, 2020-08-06:
Hello, I translated the original lasergrbl material base into French. I would like to add new materials and share them with my groups using a file how can I do I am blocked thank you

arkypita, 2020-08-29:
Material DB is not translatable.
I mean, yes you can translate but is not supported with multiple language so I cannot deploy it with new versions.

However you can find it in C:\Users\YourName\AppData\Roaming\LaserGRBL 
The file is called UserMaterial.psh and you can share this file manually

ZASto, 2020-09-05:
So, how is Material DB used in your program?
Haven't seen any place a possibility to use data from the database.

kojiro18, 2020-09-16:
Comment ajouter manuellement dans le fichier

StuartB4, 2020-09-16:
Looks like you need to have an Ortur machine to use the DB. Theres a little book icon in the top right corner of Target image settings, but for me it's greyed out so I can't change anything in there.
[image]

closed 2020-12-16 by arkypita