arkypita/LaserGRBL #1093: Custom Button Icons

tpaolillo, 2020-08-09:
Where do you guys get your custom button icons? I cant seem to even unzip the GZ files.

Thanks
Tony

StuartB4, 2020-08-10:
You don't need to unzip them. Just do this.
[image]

tpaolillo, 2020-08-10:
I know how to import them. My question was about the icons. How do I extract or get more icons.

StuartB4, 2020-08-10:
Just make your own with a graphics app, then load them in to the required button.
[image]

arkypita, 2020-08-29:
I think I have shared my icon set in some other post. Get a look here: https://github.com/arkypita/LaserGRBL/issues/306

closed 2020-12-16 by arkypita