arkypita/LaserGRBL #1095: M3 and M5 pre/post dwell command

dwburger, 2020-08-15:
I'm using LaserGRBL to drive a large (1.5m x 1.5m) drawing plotter with a servo responding to M3 and M5 commands to raise and lower the pen.  Is there a way to add a dwell command (e.g. G04 P0.3) with the M3 and M5 commands?  When the pen is lowered with the "M3 S60" command, I'd like to add a 0.3 second dwell before the pen starts to move.  This gives the servo enough time to lower the pen.  The same idea with the M5 command to raise the pen except, the dwell would precede the movement (G04 P0.3 M5).  Right now, using GRBL 0.9i, M3 and M5 are the only "Laser ON" and "Laser OFF" dropdown choices when an image is created.  Thanks!

arkypita, 2020-08-29:
For now it is only possible by exporting GCODE and manually find-replace M3/M5 with your favourite text editor (ie notepad++).

dwburger, 2020-08-29:
OK...thanks!

closed 2020-12-16 by arkypita