arkypita/LaserGRBL #1099: Custom button vars incorrect values

pelowitz, 2020-08-15:
Running in inches not mm.  Until the gcode has been sent to the cnc/laser the values in the custom button variables appear to be incorrect (possibly still in mm when should be in inches).

arkypita, 2020-08-29:
I did not understand your problem, can you make a video or picture of the issue?

pelowitz, 2020-08-29:
Easy to reproduce.  Simply create and then load into LaserGRBL a gcode file that is in inches (e.g. has a G20 instead of a G21).  At this point in time LaserGRBL (a great program by the way) has scanned the file and has set the variables used in the custom buttons accordingly.  Except, that although the values are correct in the variables LaserGRBL apparently doesn't know if they are in mm or inches.  At this point do one of the custom buttons such as "do a bounding box" and you'll witness a box in mm being traced instead of in inches as expected.  

Also, the tool tips for the Z axis always show "mm".