arkypita/LaserGRBL #1105: Laser is not working

MikeFreeroam, 2020-08-23:
Please Can you help me I build my first DIY Laser engraver... I have arduino uno with cnc shield 3.0 and I soldered everything like is in the scheme...but when I soldered ground from laser to the Gate of (mosfet) laser small beam was there and when I soldered to the Souce (mosfet) laser was on full power all of the time but when I soldered to the Drain (mosfet) laser is not working at all...I use LaserGRBL and I tried every settings 30=256,32=1... Please help me and sorry for my english :D 
[image]
[image]
[image]

StuartB4, 2020-08-23:
You can't use the SpnEn pin. If you have grbl 1.1 you need to use the Z+ or Z- pin instead. That is connected to the PWM pin 11 on the Arduino UNO.

MikeFreeroam, 2020-08-24:
Ok thanks that works for a little bit laser beam was there but now I have these problems
[image]

MikeFreeroam, 2020-08-24:
Program now is lagging like crazy...

StuartB4, 2020-08-24:
If it's not a genuine arduino you could try reinstalling the CH340 driver on the Tools menu.
Is your power supply strong enough? is it running the laser and the motors? 
Can you post a picture of your grbl configuration settings.

MikeFreeroam, 2020-08-24:
Yes I can but I tried BENBOX and in benbox it works perfectly...of course I connected pin back to spindleEn...

StuartB4, 2020-08-24:
Ok.

MikeFreeroam, 2020-08-24:
But anyway thanks for your help...

closed 2020-12-16 by arkypita