arkypita/LaserGRBL #1106: Laser doesn't turn on

DLovejoy1947, 2020-08-24:
I have a 7W laser and just purchased an EleksMaker SE v3.2 stepper motor controller board. With LaserGRBL I can move the laser along the x and y-axis, but the laser will not turn on.
Are there configuration settings or firmware that I need to change?

BeatSlayer, 2020-08-25:
Your's are...?

DLovejoy1947, 2020-08-25:
David Lovejoy
djlovejoy@woodtoysandmore.com

StuartB4, 2020-08-25:
Start LaserGRBL and click Connect then go to Grbl menu and click Grbl configuration you will get a list of settings. 
Post a screenshot of them here. You can expand the list so you get them all in the screenshot by stretching the window down a bit on the bottom edge.
Like this.

[image]

wilson740, 2020-08-27:
hola
tengo un láser de 2500 mw que no prende, tiene todo conectado, hace todos los movimientos de grabar pero No funciona el láser, no enciende, no entiendo la razón
Ayuda por favor que debo hacer
Gracias

closed 2020-09-15 by DLovejoy1947