arkypita/LaserGRBL #1107: Offline Controller

kirkbystag, 2020-08-25:
Hi

Are you able to use an offline controller with LaserGRBL? If so can anyone give advice please.

Cheers
Kirkbystag

BeatSlayer, 2020-08-25:
Save with the adv. options and copy/send to your offline controller.

regpye, 2021-03-21:
Yes you can.  I am using a LEKN offline controller with no problems.
What you have to do (very easy) is add a double pole switch to cut the TX and RT lines from the standalone controller that plug into the UNO (or whatever board you are using)  To use your computer open the switch, to use the standalone close the switch.
The standalone controller is only connected by two wires (green and yellow), plus positive 5v and ground.

alastaira, 2021-04-23:
GRBL uses a serial connection, which means you can't have both an offline controller and a USB connection to your PC active at the same time. I do the same as @regpye and have a physical switch to activate/deactivate my offline bluetooth interface (actually I just put an single inline switch in the 5V line to the controller, which has the same effect )